Builty/TonexOneController #3: Component enables in build system

Builty, 2024-11-04:
Change build system to enable enable/disable components:
- display/UI
- Bluetooth

Possibly using menuconfig to set defines like:

menu "Tonex Controller Product Configuration"

    choice TONEX_CONTROLLER_DISPLAY_TYPE
        prompt "Tonex Controller Display Type"

        config TONEX_CONTROLLER_DISPLAY_WAVESHARE_800_480
            bool "Waveshare 800x480"

        config TONEX_CONTROLLER_DISPLAY_NONE
            bool "None"
    endchoice

    choice TONEX_CONTROLLER_USE_BLUETOOTH
        prompt "Tonex Controller Bluetooth Support"

        config TONEX_CONTROLLER_BLUETOOTH_ON
            bool "Enabled"

        config TONEX_CONTROLLER_BLUETOOTH_OFF
            bool "Disabled"
    endchoice

endmenu

lcollins1989, 2024-11-04:
First of all. Outstanding work on this project! I was about to embark on similar but you’ve beaten me to it!

Quick question as I’m more familiar with Arduino/C++ style programming (self taught)

How hard would it be to add an ability to enable/disable hardware midi input using a midi cable and a standard midi input circuit (similar to the other project you mentioned in your notes) ?

Builty, 2024-11-05:
Complete

closed 2024-11-05 by Builty