Builty/TonexOneController #11: Midi commands

avlm, 2024-11-13:
Hi! I really liked the project, and planning to do something similar myself, but I'm planning to go more to a midi controller side, I not exactly need a screen, etc.

I'm planning to use this video and repository as a start point.
https://www.youtube.com/watch?v=yELnZdEJqS0
https://github.com/silveirago/OpenMIDIStomper

I know I could use the midi controller on the TonexOneController and then the TonexOneController could interface with the tonex one, but maybe this is too much things in the middle and could give me some trouble in the long term when playing live.

Since the tonexone doesn't have native midi control over usb, how could you sniff the usb commands to replicate on the esp code? could you please explain how to do that or give me a few examples of commands to send and their respective action on the tonex one?

I'm not much of a C developer, btw XD

any help would be appreciated. Congrats for making this and thanks for letting it open source! peace!

Builty, 2024-11-14:
Hi. You can see how the Tonex One commands are sent in the source file "usb_tonex_one.c"
It is quite complicated though. Its necessary to use USB "CDC" (communications device class) to send and receive serial commands to it.
You need to read the current status of the pedal, modify some values, rebuild the data again with new Crc, and send it.
This helped a lot with the understaning: https://github.com/vit3k/tonex_controller/blob/main/protocol.md

For the sniffing, you can use Wireshark on the same PC as the Tonex software, and using USB PCAP. Its hard to anaylse the data though. Much easier to copy the work of vit3k or mine.

I'm not sure if this helps at all. It won't be easy to port to Arduino. 

By the way, its possible to compile the code without the screen if you don't need it. I am working with another use to get it going on a very small $10 ESP board.

avlm, 2024-11-18:
H! thanks for answering! I'll take a look on the code and also in the option of compiling without the screen. I'll post the results here if I have success with that and put it to work!