Builty/TonexOneController #12: Hello, new Tonex FX update

waltiti17, 2024-11-14:
Hello, again thanks for all, 

with the new Tonex FX update the midi chief stop working, also I was using another App similar to Midichief and also stop working called tonexonemidicontroller.apk , the guy that develop the app will try to fix it on the weekend. the app developer guy on facebook ( https://www.facebook.com/enzo.dupont1 ) in case you want contact him. 

I waiting for the delivery of the " ESP32-S3 4.3inch " haha will arrive just in time to dont work LOL

what your future thought about your project and the new Tonex FX Update ...?

thanks

Builty, 2024-11-14:
Hi. I just fixed it. Change is committed already, I'll make new binaries soon also.

waltiti17, 2024-11-14:
you are the man !!!!!

Builty, 2024-11-17:
Fixed builds are available now. V1.0.2.2.

closed 2024-11-17 by Builty