Builty/TonexOneController #17: can't recognize chocolate plus pedal

Hank-Chung, 2024-11-22:
Thanks for wonderful project.
By the way, I have a chocolate plus pedal but it can't be connected ESP32S3-Zero. I guess there's no code for that pedal and wonder you have a plan for that pedal.
Thanks

Builty, 2024-11-22:
Hi. It should be easy enough for me to add support.
At the moment, the code is looking for the name of the device. I am looking for a better way to do this, but no solution for that yet.
If you use a phone and go to the bluetooth settings, when the Chocolate Plus is switched on, you should see the name of the device in the list. The normal Chocolate is called "FOOTCTRL" but I guess the plus is something different.
Let me know the device name, then I can add support.

Hank-Chung, 2024-11-22:
[image]

as you can see, the name of chocolate plus pedal is "FootCtrlPlus".
i'm looking forward next update. thanks.

Builty, 2024-11-22:
Code is done and committed. I'll do some new release builds in the next few days

Hank-Chung, 2024-11-22:
thanks for your support. i'll test when new update released and report.

Hank-Chung, 2024-11-28:
Today, I tested new released version and confirmed it works well with my chocolate plus pedal. 
I have two ESP32-S3 based boards which one is named zero other one is super mini, and these two boards are work fine.
Thanks for your support.

Builty, 2024-11-28:
Great, thanks for the feedback.

closed 2024-11-28 by Builty