Builty/TonexOneController #22: Waveshare Zero + MVave Chocolate latest Version not working

miasei, 2024-11-26:
Hello. Great project by the way. I could flash the latest client firmware an the ESP32 Zero, but the BT is not connecting. Tried the new beta in the Special folder. So now the Chocolate is connecting, but PC changes are not possible. It is only possible to switch from A to B on Tonex One Pedal and with a step on Chocolate it changes back to A. But nothing else works. Maybe you have help for me.

Builty, 2024-11-27:
Hi. Please set your Tonex pedal into Stomp mode. It sounds like its in A/B mode. 
The controller can only work in Stomp mode.

miasei, 2024-11-27:
Ok. My bad! So that would be great if it is so easy. Will report if it works! Thanks for the fast reply!

miasei, 2024-11-27:
Hey there again. Went home, stomp box on Tonex One and... yeah it is working! Thanks a lot again. Keep on the great work.

Builty, 2024-11-27:
Great to hear!

closed 2024-11-27 by Builty