Builty/TonexOneController #23: Not made by Waveshare ESP32-S3-zero

masaboozer, 2024-11-27:
Thanks for a very nice project.
I will report my environment.

I am using an ESP32-S3-zero board with 5V from a USB-C decoy module.
It was purchased from Amazon and does not appear to be made by Waveshare but is listed as ESP32-S3FH4R2.

I used V1.0.2.2_Waveshare_Zero_BT_client.zip and M-VAVE chocolate could not connect.
So I deleted all sources and used V1.0.3.1_Waveshare_Zero_BT_client.zip and it connected and I was able to change presets.
However, there is no sound at all.
If I press the reset button, the sound comes out for a very short moment, but then it goes silent again.
Is this due to the fact that it is not made by Waveshare?

Translated with www.DeepL.com/Translator (free version)

masaboozer, 2024-11-27:
The cable I am using to connect the Tonex One to the ESP32-S3 is an OTG compatible cable.
However, I can't hear the sound in my environment...

Builty, 2024-11-27:
I don't think a non-genuine Zero board would be a problem. As long as the chip is genuine (it sounds like it is and I've never heard of a Clone Espressif chip) then it should be OK.

Are you guys running the latest Tonex V1.8 software and have updated the Tonex One Firmware to latest?
And using Tonex One right, not the bigger Tonex?

Stomp mode is important also, as mentioned.

The cable needs to support data but that would be most anyway. There is no special cable for OTG mode.

masaboozer, 2024-11-27:
I am using Tonex One OS V1.8. Of course I am connected in stomp mode.
I will try a different USB cable later.

Builty, 2024-11-27:
Can you try doing some adjustment to the master volume and/or preset volumes, using the knob on the pedal?
It doesn't make any sense to me, but is worth a try.

masaboozer, 2024-11-28:
I tried another USB cable with the same result.
Neither the Tonex One global volume nor the preset volumes are responsive, even after installing V1.0.3.2_Waveshare_Zero_BT_client.

I checked again and I am actually sure it is Waveshare ESP32-S3-zero.

When I installed V1.0.2.2 for the first time and it did not work, I ERASE ALL with the flash tool and installed it again.
Was this a bad idea?

Builty, 2024-11-28:
The Erase All won't matter. I'm really struggling to understand why you have no volume. There have been at least 10 people in the last few days who have reported that 1.0.3.2 is working well for them.

So if you unplug the USB cable and power the Tonex some other way, it works as normal?

chrismazur, 2024-11-28:
Hmm, I wonder if the Noise Gate got set all the way up. After you change preset can you try turning that back down to 0?

masaboozer, 2024-11-28:
My setup is as follows.

Mobile battery - USB-C decoy module (converted to 5V) - ESP32-S3 - USB-C OTG cable - Tonex One

With this setup
Mobile battery - USB-C OTG cable - Tonex One
The sound comes out without any problem.

I didn't check the noise gate, so I will check it later.

I have ordered a new ESP32-S3-zero and an ESP32-S3 with display and am waiting for them to arrive.

Builty, 2024-11-28:
That power setup is different to how I think most people are doing it.
Mobile batteries sometimes have circuits that switch off the outout when the current is too low, or similar kind of things. They will also have a boost regulator to get to 5v.

Are you able to try just connecting 5 V DC plug pack to the Zero power pins? If we can remove the different part in your setup we might learn more about what is happening.
If you don't have a suitable plug pack, you could try getting a USB-A cable and cutting off one end, and solder the red and black wires to the Zero. Then use a phone charger to get the 5v.

masaboozer, 2024-11-28:
Okay, I will try another way to supply 5V.

Thank you very much.

masaboozer, 2024-12-03:
Then I installed V1.0.3.2_Waveshare_4_3_Display_Amp_Skins_BT_client when I received ESP32-S3-4.3B with display.
I was able to connect a 9V power supply and Tonex One was able to connect in stomp mode.
However, the sound comes out only for a moment and after that it is silent.
I can switch presets with Chocolate and the display also switches.

When I connect the power supply directly to Tonex One, the sound comes out without any problem.

I also confirmed that the version of Tonex One is V1.8 by connecting it to a PC.
I also updated to V1.8.1 firmware, but there is no sound.
I have disabled the noise gate and changed the preset volume and global volume, but no improvement.
What is the problem?

Builty, 2024-12-04:
This is a tough one, but I have a few more ideas.
https://www.ikmultimedia.com/faq/index.php?id=1501

The pedal has a direct monitoring setting in the PC software. Could you try to change that setting and see if it helps?

Also, if you connect the Tonex to your PC and run the software, does it also mute the Tonex analog output?

masaboozer, 2024-12-04:
Direct monitoring was set to off because the sound doubled during playback when connected to a PC...

I will try it after work!

masaboozer, 2024-12-04:
That is exactly what caused the lack of sound!

I had no problems with either with or without a monitor!
And the mobile battery converted to 5V worked fine too!

Thank you so much!

Builty, 2024-12-04:
Awesome!

closed 2024-12-04 by Builty