Builty/TonexOneController #27: Wifi/web page for config idea

Builty, 2024-11-29:
Could create a web page, hosted from the ESP32, that allows various options to be set. This would allow more customisation and less binary builds required.
ESP32 could enable access point mode (with DHCP) for e.g. 30 seconds after boot. With a password. 
Then switch it off, to make sure that someone in the crowd at a gig can't mess around.

Possible options:
- bluetooth client/server modes
- whether setting the same preset again does bypass or not
- serial/wired Midi enable
- maybe even check boxes for bluetooth client mode, to enable various pedals, and a text box for custom device name

Builty, 2024-11-30:
[image]

Wallygatorr, 2024-11-30:
It would be also great if such config menu would be available for units with screen.

Builty, 2024-11-30:
That is actually a web page. I'm working on adding this to PCBs with and without a screen, via a web browser and WiFi.

Builty, 2025-01-07:
Completed