Builty/TonexOneController #28: Support for toggling effects in ver. 1.8+

vurguuz, 2024-11-30:
With version 1.8.0 of the ToneX software there are now a number of effects available:
- delay
- chorus
- flanger
- tremolo
- phaser
- rotary

This is in addition to the previously available:
- noise gate
- compressor
- reverb.

Can these be toggled through the USB interface at all? Would it be possible to make these accessible via MIDI commands?

Builty, 2024-11-30:
Hi. As far as I know of, the Tonex One interface can't do this. They have said though that this feature is coming, so once that is done, I can certainly investigate if it can be done.

Builty, 2024-12-30:
Update here: I ended up spending about 3 days looking at how the parameters are transferred from the software to the pedal.
I've been able to read all of current values from the pedal. About 147 parameters total.

So far I've been unsuccessful in trying to send modified parameters to the pedal, but still working on it.

Builty, 2025-01-02:
Another small updateL: code relating to parameters is committed. Still not able to get the pedal to accept any parameter changes, but the code is now reading all of the parameter values and storing them locally. The 4.3" UI has the beginnings of a parameter edit screen, but incomplete.

Builty, 2025-01-24:
This is done and working now. beta version 1.0.6.1 has full control of all effects and parameters via touch screen and Midi

demym, 2025-05-23:
Hi, i am using 1.0.92, but i did not understand how to map say a CC number from Chocolate to a parameter on controller/tonex.
There is a list ? Thanks in advance