Builty/TonexOneController #29: Request: 4 foot switches on Zero

Builty, 2024-12-01:
Add support for 4 wired footswitches, which can control the presets just like the M-Vave Chocolate does. Preset 1,2,3,4. Switch 1+2 does bank down, 3+4 does bank up.

Builty, 2025-01-07:
Completed

closed 2025-01-07 by Builty

robertomoutinho, 2025-01-29:
Is the limitation on the 4.3 not having enough input ports?

The 4.3 with the edit interface and 4 foot switches would be awesome

Builty, 2025-01-29:
Yeah the 4.3 uses a lot of pins for the display, and Waveshare only allocated 2 inputs and 2 outputs on the connector.

I've had a few other requests for more footswitches, like having 6 (2 dedicated bank up/down.)
I am thinking about adding support for an I2C IO expander. For about US$6 this would add another 16 inputs. And then having the web config able to select what each input does. 
I wouldn't be able to do this for a little while though. Curently rewriting the web interface to allow full parameter control.

Builty, 2025-02-14:
Thanks.
So the web interface is finished. I do want to look at adding USB Midi support, but I will also look at IO expansion soon.