Builty/TonexOneController #31: Does this work with the TONEX original pedal?

edo44, 2024-12-06:
It would be awesome to control the original TONEX pedal via bluetooth and not use an external MIDI bluetooth receiver so I was asking if anyone had already tried.

Wallygatorr, 2024-12-06:
That is absolutely not possible, as both of Tonex pedals does not have BT. In some setups I am using Airstep controller with WIDI device connected to Tonex (or other device), sometimes chaining other devices down flow by wired midi.

edo44, 2024-12-09:
> That is absolutely not possible, as both of Tonex pedals does not have BT. In some setups I am using Airstep controller with WIDI device connected to Tonex (or other device), sometimes chaining other devices down flow by wired midi.

I already have an esp32 coded as bluetooth receiver and I'm already sending serial MIDI data to control the pedal, it would not be bad to use the USB communication instead because it is more reliable. My question was if this code can work with the ToneX big brother as well.

Builty, 2024-12-09:
I don't have the bigger Tonex to check anything, but it sounds like it doesn't have Midi over USB.

edo44, 2024-12-09:
> I don't have the bigger Tonex to check anything, but it sounds like it doesn't have Midi over USB.

[image]

Are we sure?

Builty, 2024-12-09:
Oh OK, didn't know that one.
If you wanted to send Midi over USB, it would need the Tonex One USB driver removed and replaced with a different module. A class driver for Midi.
I did try to keep host and device code separated for this kind of thing.

mchbrk, 2025-03-26:
I have bigger one too, will try then 😅

Does you @Builty thought about adding OTA? I would love to participate in the project maybe that way 😅🤓

Builty, 2025-03-27:
It probably could be done on some. Having it on only some of them makes it harder though. Documentation would need to cover both cases, and would need different partition tables for some. And some people will get confused.
I'd avoided it entirely for these reasons.

mchbrk, 2025-03-27:
I understand. Fair points somehow 😇