Builty/TonexOneController #37: ESP32-S3 1.69" Strange Noise

masaboozer, 2024-12-17:
Hi,

I am getting a strange noise (BUZZ) on my ESP32-S3 1.69”.

It works perfectly, but there is a faint strange buzzing sound coming from the ESP32.

Is there any solution to this problem?

By the way, without display (USB 5V) and 4.3” display (USB 9V) are working perfectly.

I would like to donate to your wonderful project, but PayPal does not accept donations from my country...

Builty, 2024-12-17:
Hi. It could be that the layout of the 1.69" is not quite as good as the others. I didn't notice any noise myself when I used it.
There has also been some other report it working OK and haven't mentioned the noise.

What type of power supply are you using? Could you perhaps try a different type to see if its any better?
For me, I am using a 5 volt 3 amp "plugpack". Not this exact model, but close to this one:
https://www.altronics.com.au/p/m8909d-powertran-5v-dc-3a-fixed-2.1mm-tip-positive-appliance-plugpack/

It might be better than a phone charger, but it is hard to say.

masaboozer, 2024-12-17:
I am using this USB PD decoy unit.
https://www.amazon.co.jp/dp/B09NX3F467?ref=ppx_yo2ov_dt_b_fed_asin_title

Without display and 1.69“ use this unit to get 5V. 4.3” use 9V.
The 4.3“ is 9V. The 4.3” without display and the 4.3” are working fine with no noise.

Only 1.69” has a faint buzz sound from ESP32, but it does not seem to be a problem for use.

Builty, 2024-12-17:
Another option could be to use 9V power direct to the Tonex (the power input on top of the Tonex itself.)
I haven't tried this myself, but I've seen some others doing it.
It seems that the Tonex will prefer the 9v power, if both 9v and 5v are connected.

NOTE: don't connect 9v to the USB power input!

masaboozer, 2024-12-17:
https://youtube.com/shorts/1JZXlVNk4bw?si=N0etx7DDzpDIz8EH

Builty, 2024-12-17:
Ah OK, so its not noise from the Tonex analog side. Does your PCB has a little "V2" sticker on the USB port?
Waveshare Wiki shows they changed the pin of the buzzer between V1 and V2. My development is a V2 and I am setting GPIO42 low.
I have now changed the code to set both of these pins low. Can you please test with the version attached. Thank you for finding this.

[TonexOneController_V1.0.4.1_Waveshare_1_69.zip](https://github.com/user-attachments/files/18171985/TonexOneController_V1.0.4.1_Waveshare_1_69.zip)

[image]
[image]

masaboozer, 2024-12-17:
Thanks for the response!
My board looks like V1.
I'll try it later!

masaboozer, 2024-12-18:
https://youtube.com/shorts/SbI3Y8s_CUA?si=sv4wdAspeEFTMUNn
When I tried it, it would not start.
Could you please check?

Builty, 2024-12-18:
Yeah sorry. I need to make sure I test every release, even if its very very small.
So Waveshare didn't just change the buzzer pin between version 1 and 2, they changed the entire ESP32-S3 chip to a different one. They had to move the buzzer away from GPIO 33 as that is used for the PSRAM on the V2 chip.

Im am working on this. I need to detect the chip type and only initialise that pin for the ESP32-S3R8, but not for the new ESP32-S3FH4R2.

masaboozer, 2024-12-18:
I always appreciate your prompt response!
I will be using 4.3 for a while.

Builty, 2024-12-18:
Please try this one attached.

Also, is your chip the same part number as mine?
[image]

My V2 actually has R8 on it, but doesn't have the embedded PSRAM that the R8 normally has. Too many chip versions!
I think we need to be sure the issue is V1 vs V2. 

[TonexOneController_V1.0.4.2_Waveshare_1_69.zip](https://github.com/user-attachments/files/18177836/TonexOneController_V1.0.4.2_Waveshare_1_69.zip)

masaboozer, 2024-12-18:
[image]

Builty, 2024-12-18:
OK thanks. Same chip as mine. I will keep investigating.

Builty, 2024-12-18:
I think I understand this now. So both V1 and V2 actually do use the same chip, the R8 as our photos show. The R8 has 8MB Octal PSRAM in-package.

On V1 Waveshare put the buzzer on GPIO33, which conflicts with the in-package PSRAM.
All of Waveshare code samples have the PSRAM disabled, so the conflict doesn't cause any issue, but then the 8MB ram is gone and there is not much available,
For V2, they fixed it and moved the buzzer to another pin.

In my case, I need the PSRAM for the app to function. On V2 (which I have) that's fine, but it seems like yours is a V1, and the PSRAM access is causing the buzzer to activate. I can't disable PSRAM, and I can't turn off GPIO33 (or that breaks the PSRAM and it reboots with a watchdog error, as we saw.)

I am going to contact Waveshare to see if they have any other solution.
One option could be to remove R14 or Q6, but that won't be easy as R14 is very small and both are surrounded by other components.
Blocking the hole on the buzzer might make it quieter, but won't fix it getting warm.

I'll report back when Waveshare responds.

[image]
[image]

masaboozer, 2024-12-18:
Thank you for your analysis.
I look forward to further updates!

Builty, 2024-12-20:
Waveshare support confirms my understanding is correct. There is no possible firmware work-around, if PSRAM is needed.
Removing R14 seems like the best option. Below is a diagram that shows where it is.
Can you try using a soldering iron to remove this component?

Once its removed, the buzzer is permently disabled.
Thanks.
[image]

masaboozer, 2024-12-20:
I managed to remove the R14.
It is working perfectly without any noise!

Thank you from the bottom of my heart!

Builty, 2024-12-20:
Great. I will update the documentation.

alros100, 2024-12-22:
Folks be cautious.....

I removed the resister. Have a look at my picture in case I screwed up. But look right to me

But board is no longer recognized by usb and screen shows coloured stripes. 

[image]

Al

alros100, 2024-12-22:
I deleted my last post - if anyone saw it

I can confirm the fix works perfectly.

Thanks so much Builty !

Folks be cautious with the screen. I 3d printed a case for it and it was a bit of a tight tight squeeze to fit it in. When I plugged it in,  the screen was all screwed up and I initially thought it was the resister but the unit otherwise worked. I had a spare 169 - so tried it on that and all is good.

[image]