Builty/TonexOneController #38: Uploading on Mac

DrD85, 2024-12-19:
I just wanted to share how you can upload easily on a Mac.
Just go to the website:
https://esp.huhn.me/
Use the files from the download in the right order and the right numbers and it flashes like a dream.
Maybe someone can provide the right settings, I have deleted the screenshot from the windows pc

DrD85, 2024-12-19:
[image]
Those are the files and settings you need to put into the website and make sure to use google Chrome or Edge

Builty, 2024-12-19:
Wow, thank you, this is great. I can get screen shots. I will update the instructions. So much easier than using esptool via command line.

Builty, 2024-12-20:
Docs updated. Thank you

closed 2024-12-20 by Builty