Builty/TonexOneController #43: Request: Change presets with "amp control" relays.

pk-666, 2024-12-30:
My effects switching system has 4 "amp control relays". Each relay could connect to a different GPIO, with a total of 16 combinations that could be used to select 16 different presets. That would be a killer feature.

marcuslupinus, 2024-12-30:
Its not for me to say really but I feel the project is going in all directions. 
pk-666: How many people do you think would benefit from the feature that you are suggesting ?

pk-666, 2024-12-30:
> Its not for me to say really but I feel the project is going in all directions. pk-666: How many people do you think would benefit from the feature that you are suggesting ?

I think there are far more people using switching systems with "amp control relays" than chocolate midi controllers. Even systems with a single "amp control relay" could use it to toggle between a clean and a distorted preset, exactly like most of real amps. The Tonex One works extremely well with other effect pedals and the obvious solution to integrate them are the switching systems, and almost all of them have "amp control relays".

Builty, 2024-12-30:
How would the relays be controlled? Via Midi?

pk-666, 2024-12-30:
> How would the relays be controlled? Via Midi?

I have a Boss ES-5 Switching System. It has 5 footswitches,  send/return loops for the analog pedals and "amp control" relay jacks. I can set the state of the relays for each preset assigned to the footswitches. The device has MIDI capabilities, but I don't use them.

marcuslupinus, 2024-12-30:
I don't know about amp control relay, never saw them - what is the brand/model of your device?  My ignorance.
Note that switching between two presets is standard on the ToneX One with the footswitch and the ES-5 has midi capabilities that could be used to control the Tone_X One with the project as it is I think.

pk-666, 2024-12-30:
> I don't know about amp control relay, never saw them - what is the brand/model of your device? My ignorance. Note that switching between two presets is standard on the ToneX One with the footswitch [...]

Yes, I can use the ToneX One footswitch to toggle between 2 presets, but the whole point of using a switching system is pressing a single footswitch on the controller and switch all the pedals and amp channels (or ToneX presets) at the same time.
Every multichannel amp has a jack where you plug the original footswitch to select the channel and almost every pedal switching system can be connected to that jack and act as it was the actual amp's footswitch. Both the amp's footswitch and the switching system work the same way, closing the contact on the amp's jack, grounding the circuit that changes the channel. The same thing could be done on the GPIOs of the ESP32 board.
You are right, I can order the recommended 5 pin midi board and use midi commands, but I can't find it in my country, it costs $60 to be delivered from the US, another point of faillure would be added and programming midi is more difficult.
Finally, reading the GPIOs states is the most basic thing that can be done on an ESP32 board. I'm not a programmer, but I can't imagine how that could be difficult to implement. It is as simple as reading some GPIOs states and changing the ToneX preset based on the combination of those states.

Builty, 2024-12-31:
It's certainly easy to read GPIO inputs. I just don't quite understand what the requirements are yet.
Right now I have 2 GPIO allocated for using basic dual footswitches, that do next/previous preset.

So the suggestion is to allocate 2 more inputs, and then perhaps have a mode set in the config that uses these 4 inputs to directly select 1 of 16 presets? eg  0001 for preset 1, 0010 for preset 2 etc?

pk-666, 2024-12-31:
> So the suggestion is to allocate 2 more inputs, and then perhaps have a mode set in the config that uses these 4 inputs to directly select 1 of 16 presets? eg 0001 for preset 1, 0010 for preset 2 etc?

It's exactly what I am suggesting!

Builty, 2024-12-31:
OK, got it.
There have also been requests to have 4 switch inputs, but to handle them like the M-Vave Chocolate does, where switch 1 +2 or 3+4 does a bank selection.
I can probably do both of these changes together, and in web config have a selection between dual input (next/previous), quad input (mvave style) and quad input (binary)

pk-666, 2025-01-01:
> OK, got it. There have also been requests to have 4 switch inputs, but to handle them like the M-Vave Chocolate does, where switch 1 +2 or 3+4 does a bank selection. I can probably do both of these [...]

Perfect! Thank you!

Builty, 2025-01-07:
Code changes for this binary switching scheme are complete and committed. 
I will look at doing new release builds soon.

pk-666, 2025-01-08:
> Code changes for this binary switching scheme are complete and committed. I will look at doing new release builds soon.

Man, you are a legend! I can't thank you enough!