Builty/TonexOneController #45: Serial Midi

RiccardoC58, 2024-12-31:
I have tested serial midi with a simple interface using a 6N136 optocoupler and few additional components and workds well.
On a ESP32S3 dev board, few modifications have been required, namely
Serial interface #1
input pin#17
moreover in the code the test of "C0" midi commands only works for channel#1. in order to make the channel configurable I suggest to change the code using a bitwise AND with 0xF0 to the  program change command, and extract the channel number using a bitwise AND with 0x0F. this also applies to BLE midi.

Builty, 2024-12-31:
Hi Riccardo. A few weeks ago I already made the change to midi_serial to make the channel adjustable, and mask the bits.
This is in the V1.0.4.x releases, and the channel can be set using the web config.

I'm not sure what you mean about the serial interface and pin changes. The UART allocation and pin allocations were completed at the same time as above, and are all fully coded, tested, and documented for all 3 hardware platforms.

RiccardoC58, 2025-01-01:
Good morning Greg

 

Ok for the first point, probably I have not seen the final version of the code

 

For the second point I have changed the serial port and input pint and recompiled the code to use it on the esp32S3 dev board, to avoid a conflict with the pin of the UART, since in this dev board it has a com port on serial #0 (two USB sockets). In any case the serial MIDI is perfectly working

 

 

 

You have done a very great work.

Builty, 2025-02-19:
Hi Riccardo. Thanks for the report. That's an interesting one. What I've done now is added a new config item to the web page, that 
allows the TX power to be set to 25, 50, 75, and 100%. 25% corresponding to your 28 value (7 dbm.)

I've set the default to 25% for safety, but if others want to push it harder, and if there board is OK with it, then they can crank it higher.

RiccardoC58, 2025-03-14:
Hi Greg

After some tests (and sniffing with wireshark) I have noted that in order to get automatically the stomp mode with the new version You need to add the following sentence

 

    TonexData->Message.PedalData.StateData[19] = 0x01;

 

in the function 

usb_tonex_one_set_preset_in_slot

 

 

Riccardo

Builty, 2025-03-14:
Hi Riccardo,
Thank you for the report. I have changed the code now.