Builty/TonexOneController #47: No sound output for firmeware V1.0.4.2

Gilbertnghk, 2025-01-07:
I am using s3-zero with the firmware V1.0.4.2. I can change the preset with my M-wave footcontroller, but there is no sound coming out from ToneX. Is there any settings on my Tonex OneI have to choose? I already set my toneX as Stomp mode, tried toggling on/off as well, but no help. 

The sounds can come out if I use it without any PCB. Please advise. Thanks.

Builty, 2025-01-07:
Hi. In the Tonex software, there is a setting for whether the analog output works when connected to USB. Monitor Mode or Direct Monitor or something, I can't recall the exact name.
Change that setting and it should fix your issue.

Gilbertnghk, 2025-01-07:
Thank you so much. It works now. Cheers

closed 2025-01-07 by Gilbertnghk