Builty/TonexOneController #53: Open the project to other devices

DrD85, 2025-01-24:
I love your work and I think it would be great to open it up to other modelers. One that could really profit from this would be the Nano Cortex from Neuraldsp. Is there a possibility to get the program working for the Nano, too?

Builty, 2025-01-24:
Thank you. I really have no idea about the Cortex. I does support Midi so that side at least would be easy.
A big barrier for me doing other pedals is that I don't own them. I have an old Headrush MX5 but it already has a nice UI and Midi support.

DrD85, 2025-01-24:
The Nano has Midi support, but you can just change presets. No possibility to change the effects other then Mix and no possibility to switch the effects on and off. And they have a very good App for IOS and Android for the Nano, but to change something you have to use the phone, which is not very practical live.
Same problem with the UAFX Pedals.
I don't get the aproach to use your phone to control effects. It's a nice add-on but impractical live.
Your solution opened up the Tone-X One in a big way, but if this could work with more hardware it would be the best solution for many modern digital pedals.
Thank you again for the work you did, it really is wonderful!

Builty, 2025-01-24:
Thank you. 
I suspect the manufacturers use phone interfaces because its fairly easy for them. No need to create and support another hardware device.
And not putting an LCD on the product itself drives down cost and size.

For those devices with Bluetooth, it could probably be done by connecting a ESP32-S3 to it via bluetooth an emulating the same commands that the apps are sending. It may not be super difficult.

Those that have Android apps can often be reverse engineered by decompiling the APK (eg decompiler.com.)
If they use native code you can't do much, but if its Java or Kotlin then often some good secrets get revealed.

DdMan1301, 2025-02-13:
Hi! Thanks for the great project! It's a huge job. I am using Waveshare ESP 32-S3 1.69" Display board but touch version. All functions work for me except working with a two-button footswitch. Only one button works - the next preset, the back button does not work, I used different pins to connect - it does not help. Both buttons work if you select other footswitch modes, but it doesn't work in the forward-backward mode. Is there any way to resolve this issue? I would be very grateful!!! Thanks!!!

Builty, 2025-02-14:
Hi,
The issue with the touch version is that for some reason, Waveshare choose to use 12 pins on the no-touch version connector, but only 11 pins on touch version. See below. And the pin missing is GPIO16, which is the one I just happened to allocate as footswitch 1.

Another user did a custom build to fix this. I don't have a good solution for this right now, but I can look at it. 
One option would be for me to make a platform release for that one, but its not one that I can test myself (my sample is the non-touch one.)

[image]

DdMan1301, 2025-02-14:
Thanks for the answer! Yes, I noticed the difference in contacts. Can I somehow contact the user who made the custom firmware? Or maybe I can change the parameters and recompile them myself? Or can you make corrections and I'm ready to test? Anyway, thank you!

Builty, 2025-02-14:
Leave it with me. I'll add it as a platform. Should be done later today

DdMan1301, 2025-02-15:
Thanks a lot! You're a cool guy