Builty/TonexOneController #54: Firmware Deployment Issue After V1.0.2.2 and Mitigation Method

shunpochuang, 2025-01-24:
Thanks for this amazing project!
I wanted to share my solution in case anyone else encounters the same issue.

# Issue

I own a Waveshare 4.3B model and have been unable to deploy firmware updates after the V1.0.2.2 release. The screen remains black, and neither resetting nor re-deploying resolves the issue. I downloaded the release from this [page](https://github.com/Builty/TonexOneController/tree/main/Releases) and tried deploying via both the command line and the web portal, but neither method was successful.

# Mitigation

Download the desired firmware release and replace the `partition-table.bin` file in the folder with the one from the [V1.0.2.2 release](https://github.com/Builty/TonexOneController/tree/main/Releases/V1.0.2.2). After making this change, deploy the firmware to your hardware. This approach worked for me with the latest beta releases V1.0.6.1 and V1.0.5.2.

I'm not well-versed in firmware and don't know the root cause, but perhaps recompiling the source and updating the release ZIP file could help prevent this issue in the future?

Thanks.

Builty, 2025-01-24:
Hi. Between those versions, I needed to change the partition layout. I did test upgrading from old to new partition, but I might have done a full erase.
Did you trying erasing the memory (button on the flash tool) and then uploading again?

Builty, 2025-01-27:
"0x100000 TonexController.bin"

This part is not correct now. I changed that in later versions. I needs to be 0x10000, not 0x100000.

shunpochuang, 2025-01-27:
Got it, I can upload the firmware successfully now, thanks for the help!

closed 2025-01-27 by shunpochuang