Builty/TonexOneController #57: Wired midi with a DIY midi IO board?

eniovasconcellos, 2025-01-28:
Can I build a generic midi arduino like board using optocoupler instead of adafruit? My target  is to use tonex one commanded by a roland gr55.  Can I modify de code to map the incoming midi PC from gr55 to each 20 positions of the tonex one?

Very very thanks for your project!!!

Builty, 2025-01-29:
Hi. Yes you can use a different midi board, however it needs to work with 3.3 volts. The output pin from the midi cannot be 5 volts as that could damage the ESP32.

Program change commands are already mapped to select the Tonex preset. That part might work already for you.

For Bluetooth, there are 2 modess available, selected by the web configuration.
One mode "Central" allows you to connect devices like the Chocolate pedal.
The other mode "Peripheral" is for when you want to connect to the controller from a phone/pc/tablet.
So just change the mode over to Peripheral and you should be able to pair it.
I think I remember someone else getting the Cuvave modules to work (just one of them, and paired to the controller.)

eniovasconcellos, 2025-01-29:
thank you! 

the midi input pin (5 in the zero?) needs to be pulluped by a external resistor correct?

Builty, 2025-01-29:
At the moment, PC above 20 are ignored. It would need code changes to accept an values higher than that.

Yes pin 5 is the right input for the Zero. Its a "UART" pin, serial receive. It needs to be pulled up to 3.3v, and then goes to ground briefly to follow the Midi input signal. 
The enable in wifi config controls whether the pin is set to a UART, or disabled. It doesn't make it go high or low itself (its an input pin.)

eniovasconcellos, 2025-01-29:
pulled up by software or external resistor? It seemed to me that when wired midi is set to "disable" pin 5 is LOW. And when I changed it to "enabled" it went to HIGH...

Builty, 2025-01-29:
Pullup should be internal. As far as I know of the ESP32-S3 doesn't allow a pullup to be configured on a UART pin, but based on what you are seeing maybe its doing it all by itself. It would be smart of them to do that