Builty/TonexOneController #61: MVave/Cuvave CubeTurner supported?

stereokai, 2025-02-01:
CubeTurner from MVave/Cuvave is an earlier version of the Chocolate. [10 second video showing its operation with CubeSuite](https://youtu.be/t6f3gjI74MI?si=Cxv1LGVf7epDRdgd&t=172). The replies mention the demonstrated steps worked for Chocolate as well.

Wdyt?

Builty, 2025-02-01:
It looks like it should work. You will just need to work out what the bluetooth name of the device is. For the Choc its "FootCtrl".
If its the same, it should connect OK.
If its different, you can use the web config to enable the BT custom name and enter the name.

You can view the bluetooth name using an app like "LightBlue", when the pedal is not connected to anything it should be advertising its name and LightBlue should be able to see it.

stereokai, 2025-02-01:
Thank you. Going to give it a go soon, and will return with the results for the other 1.5 people who might be interested :)

Mac-in-Memphis, 2025-03-07:
- - Thank you, Also interested. (So that makes 2.5 of us? :) Oddly, can see the CubeTurner broadcasting its device name while BT is unpaired, but it goes silent when paired via iOS settings. Can't see any changes transmitted to the iOS BluetoothLE app terminal when buttons are pressed. 
Used SincoOTA (which can see it connected) but my attempts to upgrade FootCtrl (original v30 -> v50) fails: "Upgrade file does not match device"... bought my unit third-hand, so it appears to be an Ali-knockoff. Curious 'simple' device.
Realize my above comments are cryptic, but is my 1st effort to continue this conversation ...

Builty, 2025-03-08:
The part about broadcasting (advertising) stopping when connected is normal for bluetooth peripherals. 
In order to see data from it, I don't think a terminal would be enough. It probably needs to subscribe to characteristic or similar.