Builty/TonexOneController #62: If I install a 3.7V battery to the ESP32-S3-Touch-LCD-4.3B, does USB communication not work?

joonilo, 2025-02-05:
After installing a 3.7V battery to the ESP32-S3-Touch-LCD-4.3B to eliminate noise, and connecting it to ToneX via USB-C, the ESP32-S3-Touch-LCD-4.3B shows that USB is not connected. Even when I change the patches on the ToneX, the ESP32-S3-Touch-LCD-4.3B does not change.

Builty, 2025-02-05:
A 3.7 volt battery is not high enough voltage to run USB. The USB spec is 5 volts. 
If you need to use battery power, it would be better to use a "power bank" device, that has 5 volt outputs.

It shouuldn't need a battery to remove the noise though. Were you trying to power multiple pedals from the same power supply?
The Tonex pedal works best with an "isolating" type of power supply, Or just one power supply for the Tonex, and another power supply for other pedals.

Builty, 2025-02-06:
I checked the Waveshare circuit, but nothing appears to be relevant to your problem.
I will take a guess though. The Tonex pedal needs a way of knowing if it connected to USB or not.
Some devices do that by detecting the 5v power arriving at the USB socket.
In your case, the voltage at the socket would not be 5 volts.

It might be that the Tonex thinks there is no USB host connected and therefore doesn't start up its USB receiver part.

Its hard to say for sure without having access to the Tonex schematics/code.