Builty/TonexOneController #65: Web Control

peterkummer, 2025-02-09:
Hi,
I have a Waveshare Zero and have been using TonexOneController V1.0.3.2 successfully for some time now. I updated to thw latest version and am trying to connect via webcontrol. TonexConfig is not showing on my phone when trying to connect wifi. I see ared light flashing on the Zero.
Am I doing something wrong?

andredelfini, 2025-02-09:
I've faced the same symptoms with my Zero. Haven't had time to troubleshoot it yet, as I was focused on the 1.69" display.

Builty, 2025-02-09:
For the Zero, in the latest release I had to change the partition table as the code to big for the stardard table.
I tested it on my Zero and all was OK.

Can you guys try doing a full erase, and then program again? Note also, if using the web upload method, the documentation has been updated. There is a 4th file required now, where previously there was only 3 files.

peterkummer, 2025-02-10:
After doing a full erase, I tried both upload methods. Still not working. When opening the ESP flash tool I noticed that the ota_data_initial.bin was not showing! After attaching the file and uploading still not working.

Builty, 2025-02-10:
OK, I can see why ota_data_initial was missing, sorry my mistake.
Please try this one attached. 

I tested with my Zero, doing full erase, then program, and it boots up showing 3 green flashes. TonexConfig visible WiFi and it connects to a Choc pedal.

[TonexOneController_V1.0.7.2_beta_4_Waveshare_Zero.zip](https://github.com/user-attachments/files/18731527/TonexOneController_V1.0.7.2_beta_4_Waveshare_Zero.zip)

Builty, 2025-02-11:
By the way, I think this might be same issue:
https://github.com/Builty/TonexOneController/issues/66

If the above beta 4 doesn't help, please try using V1.0.6 and see if that fixes it.

peterkummer, 2025-02-11:
I tried both V1.0.6 and V1.0.7. Get three red flashes. Reverted bak to V1.0.3. Must be something with the board.

Builty, 2025-02-11:
Thanks for the update, that's a good clue. 
Could you possibly post some close up photos of the board you have? Mainly of the chip part number.

I think are two possibilities here.
1. its just a bug that for some reason, my board doesn't suffer from. Or,
2. the chip on your board is not the same as mine, and perhaps has less flash memory or some difference in the PSRAM.

Waveshare says the chip is an ESP32S3-FH4R2 which has 4 MB flash and 2 MB PSRAM. My sampe matches this.
In recent versions the code has been growing bigger and using more PSRAM, which might explain why its gotten worse.

I'm committed to seeing this through and working out the root cause. 
After the photos, if you are prepared to, I can make a special version that dumps a log out of the USB port, so we can see why it is (presumably) crashing during the boot process.

[image]

peterkummer, 2025-02-11:
[image]

peterkummer, 2025-02-11:
Seems like the ESP32S3-FH4R2 to me. I would love to help!

Builty, 2025-02-11:
Great, thank you.
OK here is a special version. The ESP32-S3 normally outputs debug information from the USB port. I am already dumping lots of info.
When used sa a USB host however, the logging is no longer visible.
So this version has the USB host part disabled, to allow the logging to be visible. Below is what my Zero looks like on boot.

Could you please try loading this version, and then running any serial "terminal' program to view the logs. E.g. "Teraterm" on Windows. I think Mac has one as part of the OS.
You just need to connect to the same COM port number as the programmer app does, and use the settings of baud rate 115200, 8 data bits, 1 stop bit, no parity.

After the program update you will need to press reset button or unplug/replug to get it out of programming mode.
Then hopefully you can catch the logs. You might need to again press reset button with the terminal program open to see them.

Please post up what the logs show. I'm guessing there is a "guru meditation error" (that's a real thing!) or similar kernel panic that makes it reboot over and over.
Thanks for the help.

[TonexOneController_Zero_noUSB_terminal_V1.0.7.2.zip](https://github.com/user-attachments/files/18747292/TonexOneController_Zero_noUSB_terminal_V1.0.7.2.zip)

[image]

peterkummer, 2025-02-11:
I hope this is what you need.......

[MyZeroLog.txt](https://github.com/user-attachments/files/18748153/MyZeroLog.txt)

Builty, 2025-02-11:
Yes thats exactly it. However it doesn't show any errors. Did you get 3 green flashes from the onboard led?

Please repeat the same test again with this one. It has less removed, plus some memory dumping. it still won't run the Tonex yet though.

[TonexOneController_Zero_noUSB_terminal_V1.0.7.2-2.zip](https://github.com/user-attachments/files/18748448/TonexOneController_Zero_noUSB_terminal_V1.0.7.2-2.zip)

peterkummer, 2025-02-11:
no three quick red flashes and the constant red flash. Will get back to you!

peterkummer, 2025-02-11:
[New Text Document.txt](https://github.com/user-attachments/files/18750488/New.Text.Document.txt)

DrD85, 2025-02-11:
I've set the web-Tool to stationary and wrote my credentials in the mask, but something must have been wrong, cause it won't connect to my wifi. How can I access it now?

Builty, 2025-02-11:
For DrD85, you can do a "factory reset" by pressing and holding button 1 for 15 seconds. Button 1 is the first wired footswitch as per https://github.com/Builty/TonexOneController/blob/main/HardwarePlatforms.md#footswitches

If you don't use this normally, you can just link the pin to ground temporarily.
This procedure will revert back to access point mode with default credentials.

Builty, 2025-02-11:
For Peter, yeah OK that log does indeed show it boot looping. Very interesting that it resets at random points during the boot cycle, and no cause is shown. I.e.. it doesn't appear to be a software crash.

The blue led is interesting too. I'm not coding it to show blue at all, but there are various led models that can use either RGB or GRB colour order. So I'm wondering if this led is relevant.
I added that led flash in V1.0.5.2.

When I get home from work I will make another test build, but if you have time would you mind checking if V1.0.4.2 and V1.0.5.2 work OK on your board?
If V1.0.5.2 has the same issue, that would really point to that led, as not much else was changed betweeb 1.0.4.2 and 1.0.5.2.

DrD85, 2025-02-12:
Link The ground from the footswitch to Do1? I have the 4.3 Version

Builty, 2025-02-12:
For the 4.3B, DI0 to ground. Or just erase the flash and program it again

Builty, 2025-02-12:
Peter, if you could possibly try V1.0.4.2 and V1.0.5.2 and let me know which if any work, that would be another nice clue.

Also below is beta 5. Based on the fact that the led on yours (and a few others) is flashing a different colour, I'm thinking these boards may have a different model led fitted. The genuine World Semi WS2812 is GRB colour order (matches mine) but these appear to be RGB and not a genuine World Semi. I've been burnt a few times in the past with this kind of thing.

So this version has the colour changed to blue, which should always be blue regardless of led brand. And the brightness is reduced from 100% right down to 5%, on the theory that max brightness might be drawing too much current and causing the power supply to brown out.

Could you please also try this one. Its full function, not a cutdown debug version.
Thanks.
[TonexOneController_V1.0.7.2_beta_5_Waveshare_Zero.zip](https://github.com/user-attachments/files/18764609/TonexOneController_V1.0.7.2_beta_5_Waveshare_Zero.zip)

peterkummer, 2025-02-12:
Hi,
I found out, that if I left the board connected to the PC through the usb port, everything works fine (apart from the led flashing red)! As soon as I connect to an external powersupply no success apart from V1.0.3. I tested all versions.
I am going to try out another power source and will get back to you.

jesusglezalv, 2025-02-13:
Not sure if it's the same issue here. I'm able to flash my zero with beta 5, it boots flashing in blue three times. I quickly connect to wifi and open tonex.local. 

,It shows the nice new web config, but just after a few seconds (in total since I boot it up it was much less than 60 seconds) I get a popup window "Waiting for Connection to Controller..."

I've tested with two Zero boards and in both cases happens the same.

Other than that the controller seems to work right (tested with my chocolate BT).

In my case the controller seems to be different to @peterkummer 's one (both Zeros has exactly the same numbering in my case. Don't know if that's related to manufacturing batch, date....)

[image]

Builty, 2025-02-14:
With the "waiting for connection.." thing, it works like this:
- the web page is delivered to the browser
- Javascript in the browser uses "web sockets" to make a connection back to the controller, which is then used to send/receive data without having to refresh the web page. The message you see is when waiting for this secondary connection

What browser/phone are you using? Do you have any way of viewing the console output from the Javascript? E.g. on the Chrome web broswer, in the development tools there is a console viewer. 
I'm writing some debug info here that might help.

Maybe also some browsers have a security setting or something that enables Javascript or local connections or something?

peterkummer, 2025-02-14:
I installed the latest version on my brand new Zero........three blue flashes and wifi connection working!!!! I must have fried the old board. Sorry for wasting your time!

Builty, 2025-02-14:
No problems at all. Great that you have it working.

peterkummer, 2025-02-15:
Hi,
About the buttons in webcontrol with '??'. I found out, that on the TonexOne the second(most right) one toggles the cab on/off.

Builty, 2025-02-16:
Thanks for the report. Yeah I'm struggling to figure out what is happening with this switch. I've seen it reject the change and revert back to being on.
I'm hoping the IK update will reveal a few more secrets here.

peterkummer, 2025-02-17:
Do you know what the other two buttons do? Soundwise something happens....

Builty, 2025-02-17:
I haven't been able to figure it out yet.

peterkummer, 2025-03-26:
Hi Greg,

Version 1.0.8.2 Beta 6 is working as it should on my Waveshare 1.69. But I still have problems with webcontrol. It connects perfectly the first time, but after disconnecting after a couple of minutes, when trying to reconnect, TonexConfig is not showing anymore. This is after changing from Access Point Timed to Access Point.

Peter

Builty, 2025-03-27:
Hi Peter. I ran mine for about 10 hours today. It didn't do anything unusual.
I'll try some more tests with connect/disconnect/connect etc