Builty/TonexOneController #66: I cant make my TonexOne turn on

Alfrednajera, 2025-02-10:
Hello I just received my esp23-s3 and I did the boot thing just as the video, but when I try to turn on my tonex it just doesn't have the enough power to turn on, I tried the last firmware and nothing works, im 100% that its the enough power, Im a total beginner with this, I´ll appreciate all the feedback <3 :)

Builty, 2025-02-10:
Hi.
Could you provide some more details about your setup:
- which PCB are you using? 
- how are you providing power? Where is it connected, and what is the voltage and current of the power supply?
- have you tried with an older firmware? I've had some reports of issues with the Zero board on V1.0.7.2 that I am working on

Alfrednajera, 2025-02-10:
If what you mean with the pbc is what waveshare product im using its the zero, and nope I haven't tried the other firmware, it is connected to a protoboard, there I have the jumpers connected to the dc adaptor, there the dc to usb cable and finally either a usb port from my Mac or a normal phone charger and nothing

Builty, 2025-02-11:
Could you try this test please. Disconnect the Tonex, Connect a USB-C cable to the Zero, and the other end to the phone charger and switch on.
Does the Zero board give 3 flashes of the green led?

If it doesn't, please try to load version 1.0.6, and repeat the same test. Do a full erase before loading 1.0.6.

I think we need to make sure the Zero is booting up normally and running, before we introduce the Tonex pedal.
Thanks.

Builty, 2025-02-11:
With the above test, I mean with the Zero taken out of any protoboard, so nothing else is connected to its pins. Just the USB-C by itself

Alfrednajera, 2025-02-11:
Okey, yes the zero doesn't give me any lights at all, how could I do the full erase of the firmware? sorry that I don't know anything I just try no to mess things up

Alfrednajera, 2025-02-11:
nevermind I did it with the terminal on my Mac, let me download the firmware and try everything again

Builty, 2025-02-11:
You beat me to it.

Erase button here.
[image]

Also could you post a screen shot of this part of the app:

[image]

The reported info might give me some clues

Alfrednajera, 2025-02-11:
Perfect let me do it

Alfrednajera, 2025-02-12:
Okey I did it, I erased the data from the zero and load the 1.0.6 firmware, the zero when its connected just via usb-c it does turn the 3 green lights and even connect to the chocolate plus but it cant turn on the midi, if I try powering the tonex one with the dc cable and the usb to the zero the chocolate cant change the presets, im un the custom mode from the chocolate btw, ill send you the screenshot 

[image]

Alfrednajera, 2025-02-12:
I tried the other modes from the chocolate and nothing as well

Builty, 2025-02-12:
Its not easy to understand what is happening here. Its a bit easier on the boards with LCD displays.
So this could be an issue with the communications with the Tonex pedal, or it could be an issue on the Chocolate site.

I've been dealing with other people having issues on the Waveshare Zero (see above link) but did you try V1.0.7.2 already?
If you can get your board to boot on V1.0.7.2, you could connect to the WiFi, and use the web page to see if you can change presets from there.
Refer here for more details on the web control:
https://github.com/Builty/TonexOneController/blob/main/WebConfiguration.md

If it doesn't change presets from the web controller, then its seems like an issue with communications to the Tonex.
For USB-C cables, so cables do charging only and won't do communications. Other cables can do both.
Make sure your cable to the Tonex can do communications. You can test by using it to connect e.g. a phone to a PC and make sure it appears on the PC.

With the Chocolate pedal, the configuration should be the default mode where it sends program change commands.

I think these tests should help us work out where the problem is.

Alfrednajera, 2025-02-12:
Okey, I booted the zero with the 1.0.7.2 and the midi controller works correctly to change presets but its very unstable, sometimes works correctly and sometimes the tonex stop responding (that happens most of the times) but now we know that there is no issue with the chocolate or the cable, since what release I can use the chocolate plus?

Builty, 2025-02-12:
Chocolate Plus support was added in V1.0.3.2.

Now I've been dealing with a number of users with issues. It's looking the problems they have been having are related to using power supplies that are too small. Particularly 5v phone chargers that only give out 0.5 amps.
You might want to try a bigger power supply, to see if that fixes the stability issues.

Alfrednajera, 2025-02-13:
Okey, I managed to make everything work well and no issues but the audio when the zero its connected its disabled, I mean if I unplug de usb c the sound of the pedal come back, that has to do with the fact that I am powering the pedal with the 9v and the zero simultaneously ?

Builty, 2025-02-13:
The audio one is a setting in the Tonex Software. I think it called "Direct Monitoring".
It controls if the analog sound is muted when using USB.
Once you change that setting, all should be good.

Alfrednajera, 2025-02-13:
okey, I used a 20w usb charger and nothing, the tonex still can't turn on, how do you manage to make everything work with jus one power supply?

Builty, 2025-02-14:
I have another Zero user who has much the same isssue as you. He is also using a phone charger.
For me, I use a "plug pack" (maybe called "wall wort" in some parts of the world.) It works well.

I was discussing with this other user about phone chargers and the PD (power delivery) system they use. They will often start with 5v and lower current, and the phone and charger talk over USB to work out what the phone can handle, and then the charger gives it more voltage and/or current.
It might be that your charger is 20 watts only when the phone has asked for it. Those negotiations can't occur on the controller.
Some chargers may also look at the USB pins and check resistance etc. That can't happen if its just 2 wires connected to the controller and other wires cut off.

Would you have some kind of "dumb" 5 volt supply that you could try? Not a phone charger as such, but something that just gives 5 volts 1 amp or higher regardless of what is connected.

I think I might need to update the documentation to give warnings about phone chargers, but I need to make sure this is really the issue.

[image]

Builty, 2025-02-14:
More: yes it seems some chargers look at the resistance on the data pins to work out the maximum current:

[image]

Builty, 2025-02-14:
And also this: https://forum.contextualelectronics.com/t/can-you-pull-current-from-usb-c-5v-power-without-negotiation/3577

Alfrednajera, 2025-02-17:
I haven't wrote anything because I haven't had the chance to buy one of those but ill write the update as soon as I can get one.

Alfrednajera, 2025-02-23:
Finally it arrived and it works amazing, I'm using the long/short step and it works really good, I couldn't be more satisfied with how it works!

closed 2025-02-23 by Alfrednajera