Builty/TonexOneController #71: Possible issue with PC messages

alros100, 2025-02-14:
Hi Builty

I wonder if this is something unique to my setup (using cme widi group), but I just noticed something weird.

My ToneXone on 169 board.is set to midi channel 1. It responds correctly and I am currently just using the cc messages to change to next / previous preset and for controlling effects. However, while setting up a looper on iPad (ch4 pc changes 1-4), I noticed that the tonexone changes to the corresponding preset. I tried two different foot controllers and sending pcs on different channels and it seems that tonexone responds to all irrespective of the channel.

Is this by design ? I tried disabling chocolate and XVive with no difference in behaviour

I was on 1.6 beta 3 and just flashed the 1.7 beta 5 and it shows same behaviour

Thanks
Al

Builty, 2025-02-14:
Thanks Al.
Yeah OK this is a bug. I'm not checking the Midi channel for Bluetooth like I am for wired Midi. 

I've fixed it now, look out for beta 6 later today.

alros100, 2025-02-15:
Amazing ! Thanks so much !

Al

Builty, 2025-02-15:
Released to beta now.

alros100, 2025-02-15:
Thanks So much Builty! That looks to have fixed the issue. 

Thanks
Al

rhcpschipper, 2025-04-02:
> Thanks Al.
> Yeah OK this is a bug. I'm not checking the Midi channel for Bluetooth like I am for wired Midi. 
> 
> I've fixed it now, look out for beta 6 later today.

So when setting Midi channel on the web editor it also corresponds to the BT midi?