Builty/TonexOneController #90: Control 2 Tonex Ones with same footswitch

johnk877, 2025-03-03:
I have 2 Tonex Ones on my board, and I want to try controlling them in tandem with 1 set of footswitches. I have some success with wired controllers -> 2 waveshare zeros -> 2 Tonex Ones.

But what if I want to control both in tandem using bluetooth? Any ideas?

Builty, 2025-03-03:
Hi. I have another user who has the same request. Bluetooth doesn't allow connection of one peripheral device to multiple central devices at the same time. 
I don't have a solution for this at the moment. It won't be possible with bluetooth but could be via some other link between the controllers.

The other user is looking at using wired Midi and splitting the same Midi signal to both controllers. 

I'll keep thinking about what other options there could be.

pwaugh, 2025-03-03:
Or WIDISent from my iPhone

closed 2025-03-05 by johnk877

DZYHRY, 2025-05-14:
Hi @johnk877 , how did you control I have 2 Tonex Ones with 1 set of footswitches (wired) -> 2 waveshare zeros -> 2 Tonex Ones.

Can you please enlighten me how did you avoid short circuiting the two ESP32 S3's? Can you share you wiring diagram please.

Thank you.

johnk877, 2025-05-20:
Hmm I just connected the same set of buttons to both controllers. Nothing shorted