Builty/TonexOneController #93: Waveshare zero intermittent wifi on latest version

johnk877, 2025-03-05:
I am facing intermittent wifi issues with my Waveshare zero boards (both boards) with the latest firmware. Downgraded to 1.06 and seems to work fine for now. It worked fine when I last connected the boards a week ago, but suddenly the wifi is intermittent. Reflashind doesn't help either. Any advice?

Builty, 2025-03-05:
Hi. In the last few weeks I've identified a few issues in this area.
One user reported that with the WiFi transmit power set to maximum (the default) then his board WiFi was not very reliable.
I have now added a transmit power control setting, and the new default is 25%. This fixed the issue for that user (sorry I can't remember the user's name, but its in the Github issues area.)

Secondly, this week I found a memory leak in the web interface. If used for a while, the micro would run out of memory and then malfunction. This is fixed now also.

These fixes will come with V1.0.8.2. I am hoping to finish testing this week and release it then. If you really want to try the fixes before that I can post a pre-release build here.

johnk877, 2025-03-05:
Thanks! You're a champ!

Builty, 2025-03-06:
OK, check this one out. Please give feedback on if it helps. Thanks.

[TonexOneController_V1.0.8.2_preview_2_Waveshare_Zero.zip](https://github.com/user-attachments/files/19099827/TonexOneController_V1.0.8.2_preview_2_Waveshare_Zero.zip)