Builty/TonexOneController #97: Send midi Messages without the extra board

DrD85, 2025-03-09:
Hi @Builty 
is it possible to connect a "normal" auxswitch to the 4,3 board and toggle effects on and off, like its done with the new external SX1509 IO?
So that I just connect the two button auxswitch and toggle compressor and Modulation with it?

Builty, 2025-03-10:
Hi. At the moment no, but that sounds like a good request. 
I have release V1.0.8.2 but I am expecting other requests in the footswitch area, so I can look at these soon.

SergyLo, 2025-03-10:
Hello. I put together such a project on the esp 32-s3 zero board, it does not work with the BJ Devices tb-11 foot controller via the midi cable. It works well with m-vave chocolate plus V9 via cable.

SergyLo, 2025-03-10:
Hello @Builty 
I put together such a project on the esp 32-s3 zero board, it does not work with the BJ Devices tb-11 foot controller via the midi cable. It works well with m-vave chocolate plus V9 via cable.

DrD85, 2025-03-10:
some kind of visual representation, when a effect is turned on would be great, too. Thank you again, for the great work!

Builty, 2025-03-10:
> Hello [@Builty](https://github.com/Builty) I put together such a project on the esp 32-s3 zero board, it does not work with the BJ Devices tb-11 foot controller via the midi cable. It works well with m-vave chocolate plus V9 via cable.

Hi. I don't see any reason why it wouldn't work. Did you check the settings of the tb-11? Make sure its Midi channel matches the controller's Midi channel. 
Would you some kind of Midi interface on your PC? e might be able to check exactly what data is coming out of the tb-11 to see what is wrong.

Builty, 2025-03-10:
> some kind of visual representation, when a effect is turned on would be great, too. Thank you again, for the great work!

Yeah another user has changed to an LCD display and added labels for the effect status. I like this idea and will see what I can do

Builty, 2025-03-14:
In the current controller firmware, I have to do a trick where I read a huge chunk of data (about 30 kbytes), change a few bytes, and then write the entire large chunk back again. It was the only way to make it work before. It was not very fast to do this process.

Now that IK has released the editor, they have changed the pedal firmware to allow the sending of just one parameter value, by sending a small message. Yesterday I changed the controller code to use these new messages (I was able to reverse engineer them easily due to the detailed work I had done before on the parameters.)
I have it working now, and its much faster. Code is simpler, faster, and uses much less memory also.

I will release beta 3 with this change in the next day or 2. I just need to test some more to make sure I didn't miss anything.

Builty, 2025-03-14:
Note also, I found a bug in the new expansion footswitches. For example, if you are on preset 1 and enable delay, but then change to preset 2, when you press the footswitch again delay would be commanded off. If however preset 2 already had delay disabled, then this footswitch press does nothing. You would have to press the footswitch a second time to enable it again.

I think I need to read the current state of the preset effect, and change the footswitch toggle so that it always does the opposite of the current preset setting.

DrD85, 2025-03-15:
If you need somebody to help you test new features, I'm always interested!

DrD85, 2025-03-15:
And is there a possibility to change the bank up and down settings? I've built my control unit but channel 1(2 on the app) isn't working. Now I can go only up, but not down again.

Builty, 2025-03-15:
DrD85 yeah if you could help test this beta 3 build, it would be appreciated. Attached below.
The icons on the front screen of the 4.3B can now be pressed to toggle the item, and the effect and preset switching should be faster.

For the bank up/down, anything is possible but I need to try to keep the configuration not overly complicated. Is it possible to fix yours? Can you suggest a way of configuring the buttons?
This whole area needs work. I've had requests to allow effect switching from the 4 directly connected switches. And also to get tap tempo in there somewhere. Plus possible options of having the preset switching work when button is pressed instead of on release, if no banking is used on the switch.

I'm open to ideas on this one.

[TonexOneController_V1.0.8.2_beta_3_Waveshare_4_3B_Display_Amp_Skins.zip](https://github.com/user-attachments/files/19259896/TonexOneController_V1.0.8.2_beta_3_Waveshare_4_3B_Display_Amp_Skins.zip)

DrD85, 2025-03-15:
Maybe do it like the FX-Buttons. So that the Footswitchtes are assignable, juts like the fx.
Button 1 does x. Bank up down is button x + button y

DrD85, 2025-03-15:
Should I write problems, that I've found directly here?

DrD85, 2025-03-15:
First impression: Wow, switching presets off is now much faster and there is no gap.

Builty, 2025-03-15:
Yeah here is OK for this one. As the change was quite a significant "structural" one I think the additonal testing is needed.
Sounds like we might be on a winner.

DrD85, 2025-03-15:
Absolutely. I played with it for around an hour now. What a joy and no problems so far. I had one short crash when switching the reverb on and off, but I couldn't replicate it so far. The biggest issue is that the foot switches sometimes need extra clicking, because the don't know if the effect is on or off. So if a change preset and the foot switch toggles to off, it need two clicks to activate the effect. Other than that, it works really great. If I find a problem, I'll post here, but it looks like a winner, indeed.

Builty, 2025-03-15:
Awesome. I had a good play with the version last night also, and yeah the effect switching speed is magic.

Yes I mentioned that footswitch bug above. I'm working on it.
Thanks for helping out.

Builty, 2025-03-16:
Regarding footswitch toggle issue, its fixed now. For "boolean" (on/off) parameters, I get the current value first, and then flip it to the opposite one. 
Beta 4 coming shortly

DrD85, 2025-03-16:
Wow, that is great news. If, it's possible to assign the Buttons 1-5 free and set a button combination for Bank up and down, I'm in heaven.

Builty, 2025-03-17:
Yep, I've added another 6t foot switch configs. One of which is "1X7B" which is 7 total switches, 5 for presets and 2 dedicated bank buttons (the last 2.)

Will come with the next releases, not too far away, Trying to do tap tempo first,

DrD85, 2025-03-30:
I noticed that the problem, that sometimes a double tap is needed still occurs, when changing Reverb Types. 
I programmed one fx to switch between a spring and a room Reverb and there I have to double tap sometimes. Also an indicator for switching reverb types would be great. Maybe a blue and a purple Reverb logo, if that's possible.

Builty, 2025-03-31:
Thanks, yeah I can see why. I need to find a more general fix that takes into account the current parameter state when toggling.
I have noted it to fix.

DrD85, 2025-04-18:
With 1.0.9.2 the toggling between reverbs seems to be broken? I tried to switch between spring and plate, but it just switches, if I select another reverb type, than it switches to the first selected reverb type, but toggling is not working right now. Also would it be possible to add letters to the reverbsymbol to show, which reverb is active?
Besides this bug 1.0.9 is a great update!!!

Builty, 2025-04-18:
Yeah I had another report of similar. I fixed that "sometimes a double tap is needed still occurs" bug, but seems I've made it worse. I will look at it today.
I added letters to the modulation symbols yesterday. I will look at the reverb symbols also, won't be hard

Builty, 2025-04-19:
Please try with beta 2.

DrD85, 2025-04-22:
Tested beta 3 today and so far nor problems with toggling the effect. It's really impressive how good the software is.

closed 2025-04-22 by DrD85