Builty/TonexOneController #100: No connection -  after instalation new Tonex Editor  and Tonex One firmware update to version 1.3.11

Przester-MZ, 2025-03-13:
Hi Builty 

First of all, big thank you for this project and your work !

Unfortunately after today installation  of new Tonex Editor and update Tonex one  firmware   to  version 1.3.11 my Waveshare ESP32-S3 Zero with beta 1.0.8.2 firmware stop communicate with tonex one. I can connect to web configuration or my M-wave Chocolate foot-switch seems to be connected via bluetooth to Waveshare as usually but it looks like there is no connection between tonex and a zero platform. 
I don't know if I need to set something in tonex software and this is only my problem or they change way of communication in general in this firmware?
Is there any option to downgrade tonex one to previous firmware ?
Any advice ?

peterkummer, 2025-03-13:
I have the same problem.

Builty, 2025-03-13:
Hi Guys. Yeah IK made a little change that broke things. Not sure why they did it.
V1.0.8.2 beta 2 is released now with the fix.

peterkummer, 2025-03-13:
That did it! Thanks for the quick response!

Builty, 2025-03-13:
Great. I was half expecting that this would happen. I'd been holding off on making large changes to make sure the code base was in a good stable state and ready. Cost me a few hours late last night but thankfully the break was very small and easy fixed.

Przester-MZ, 2025-03-14:
It is working again.  You are so quick in response - Thank You.

tangoprimero, 2025-03-15:
After installing the latest update, I had the same issues that others had. There was no communication. Then I installed the latest update (beta 2). Now there is communication with Tonex-One, but it’s not working properly. All four buttons of Chocolade make Tonex-One do the same thing, exactly the same, regardless of how they were programmed with CubeSuite, meaning: They all change a specific sound to another, but only once, and in one direction, and with any of the four buttons. If I change the sound with the Tonex-One button, I can change it again with Chocolade, but only once, only a specific sound, and only in one direction, and so on.

tangoprimero, 2025-03-15:
I forgot to write that I updated the Tonex One firmware to version 1.3.11 and I'm using the ESP32-S3 Zero with version 1.0.8.2 (beta 2).

Builty, 2025-03-15:
I have another user with the same issue. Can you try manually enabling Stomp mode on the pedal.
I will have beta 3 coming soon, which has another fix in this area.

tangoprimero, 2025-03-16:
If I enter "stomp mode" manually, as you recommended, everything works correctly. I'm looking forward to "beta 3." Thank you very much for your work.

Builty, 2025-03-16:
OK great. Another user was kind of enough to work out what was needed to enable stomp mode in the communications, so beta 3 hopefully will do this automatically and will just work. Beta 3 coming in about 30 minutes from now

corradoricuccieng, 2025-03-17:
I had the same issue since update Tonex one firmware to version 1.3.11, i upgraded the ESP32-S3 Zero with version 1.0.8.2 (beta 4) and it works! Thanks a lot!

dezindocavaco-create, 2026-11-04:
estou com meu tonex one na versão 1.3.17, e com uma placa ESP DevKitC N16R8, e mesmo com a instalação do firmware do Tonex controlar V2.5.0.2 não consigo gerar a comunicação do chocolate para o tonex a placa reconhece o chocolate mas nao reconhece o tonex.

Builty, 2026-11-04:
Hi.
The most common causes of this are:
- using a USB-C cable that is for charging only. It must be one that supports data also
- make sure you use the right USB-C port, the DevKit-C has 2, see below
- make sure you solder the little "OTA" pad on the back of the PCB, see below also

[image]

dezindocavaco-create, 2026-11-04:
Muito obrigado, funcionou era o link OTG USB, que estava faltando, muito grato pela sua atenção.