Builty/TonexOneController #103: Noisy signal from IUSB

buzzyc, 2025-03-16:
Hi Builty, 
Not sure if anyone else has reported or experienced this, but I am difficulty getting a clean signal to any of my amps once I plug a device into the Tonex.
The sound is a combination of hiss and a ticking pulse (almost like a clocking signal)
I have tried the following with little to no success:

- 4.3B connected to its own power supply
- 4.3B connected to same 1spot power supply
- 4.3B connected to isolated power supply
- 4.3B connected to 9v Lithium battery (via USB, not battery connectors)
- 4.3B at a different house to remove my own grid as an introducer of noisy power
- 1.69 connected to standalone bench power supply

I have tried multiple USB cables
I have tried different amps
The problem always persists
I don't need to plug the device in all the way sometimes - at times just the ground planes on the USB cable shielding making contact with the Tonex One is enough to set it off.

I have powered the Tonex and the 4.3B or 1.69 individually instead of relying on passthrough power, and whilst the noise quietens, it is still there.
The biggest issue I find is the noise is not consistent - sometimes it disappears - and it trying to replicate the times it isn't there, suddenly the noise returns

Using the Tonex One via USB to say my laptop or a USB midi pedal does not introduce this same noise, when trying to replicate if the Tonx One USB was faulty.

To be honest, I don't necessarily expect a solution - rather if you or anyone else has experienced this?
It's doing my head in on an otherwise perfect project.

Cheers, 
Lee

DrD85, 2025-03-16:
Did you try a usb battery pack? I use them exclusively with my board and I have no noise at all.

pwaugh, 2025-03-16:
You need a higher quality USB cable.Sent from my iPhone

buzzyc, 2025-03-16:
I have tried with a USB battery pack as per my list of troubleshooting.
I have high quality USB cables.

pwaugh, 2025-03-16:
Have you switched cables?  What cable are you using?Sent from my iPhone

Builty, 2025-03-16:
Hi Lee,
A few people have had mild noise issues, but nothing like you describe. Some of us (including me) have noticed that if you power another pedal from the same power supply, it often gets noisy. An isolated supply works best to avoid that one.

What do you have connected to the Tonex output? An FRFR? Have you tried connecting headphones instead, to see if the noise is still there? 
A few users have been powering the Tonex off the 9v DC jack on it, and connecting the USB just for communications. That could be worth a test also.

buzzyc, 2025-03-16:
Hi Builty,

I have tried an isolated power supply & powered the Tonex separately as per my inital post.
Whilst the noise gets quieter, it does not go away - primarily the pulsing noise persists, just at a lower volume..

I generally run the tonex through the effects return of a combo, but the noise can be replicated on another amp and through my audio interface.

I will try headphones.
One test I did last night was plug my phone in over the same USB cable and run midi chief - that worked fine without any noise introduced.
I also found I could play music through the tonex from my phone in this manner - again with no additional noise introduced into the signal chain.

So process of elimination suggests it is the ESPs, but happening on both the 4.3 and 1.69 is annoying, given I bought the 1.69 to try and work around what I thought was a 4.3 specific issue. :(

buzzyc, 2025-03-18:
Well, the noise has disappeared for now
I ended up soldering a spare USB C breakout board I had to some dupont wires so I could 'breadboard' the power with the wires coming off the 1.69
Using the same USB cable, same power as before, the noise is now completely gone.
I will do more testing with the 4.3 as well but for now - I am happy to have some success so I can focus more on playing than troubleshooting ha.

Builty, 2025-03-18:
Good news. Mind sharing a photo? It might be helpful if others experience the same issue. Thanks.

pwaugh, 2025-03-18:
Try a different length USB cable too when you get a chance, but one the is not a multiple of the current length. Sent from my iPhone

LouisHrg, 2025-05-28:
Maybe I'm late to the battle, but I have the same exact problem on my setup (waveshare zero), I cannot use the tonexonecontroller as the same time as my other pedals (I use myvolt usb cable to power the board and my other pedals). I have a bad and loud hiss coming from the tonex.

Builty, 2025-05-29:
The pedal really doesn't seem to like being on the same power supply as other pedals. 
It works better with isolated supplies, or using a separate plug pack for the Tonex.