Builty/TonexOneController #105: Bluetooth issues with latest beta

faso82, 2025-03-18:
Bluetooth connection between M-Vave Chocolate Plus and ESP32-S3 is not stable with the last beta release. Anyone else noticed the same issue? Any suggestion?

Builty, 2025-03-18:
Hi. I can't think of any reason why it would have gotten worse from the code.
You could roll back to an older version to double check that its not something else.

I have noticed that the Chocolate (not plus) uses a tiny antenna on its PCB, which is inside a metal box. It has quite low signal strength. Maybe the Plus is similar?
Does it get better if you are close to it?

You can see this with an app like "LightBlue" when the Chocolate is not connected. Signal strength when phone is right next to it is -57 dbm. My PC with Bluetooth is further away and reads -36 dbm.

From 1 meter away, its about -75 dbm, which is considered quite low. 
https://www.metageek.com/training/resources/understanding-rssi/

Also make sure there are no metal objects near the ESP32 antenna chip, and that its not in a metal box itself.

[image]

Przester-MZ, 2025-03-18:
as Greg mention Chocolate and Chocolate plus has not the best construction due to metal box and small antenna inside  but I made a upgrade from this video and it improved a lot a range of connection. [https://www.youtube.com/watch?v=ToILhTZ5NZc]

buzzyc, 2025-03-18:
I noticed it would be less responsive if the battery on the chocolate got low.
Charging it to full battery would fix the intermittent issues.
I just assumed as the battery wore down the transmitter became weaker.
I'll look at that tutorial to improve signal too - thanks!

faso82, 2025-03-18:
thanks everybody for the replies.
as soon as i can i'll try to recharghe the battery, and replicate the conditions where i've had some issues
then, if it doesn't work, will downgrade the firmware and doublecheck it
the Przester video is super clever, will check out the solution soon

50pln, 2025-03-19:
My modification (as in the video). I left the antenna vertical, it improved the range on the floor.

[image]