Builty/TonexOneController #106: Help me svp

sebaone84, 2025-03-19:
Hello,
I have the Tonex One with the 4.3" ESP32S3, and I have a Chocolate and a Chocolate Plus.
 Is it possible to connect them at the same time?
 So I can have one to manage the banks and one for the effects?
Thank you very much for your time and work!

Builty, 2025-03-19:
Hi. At the moment, only one bluetooth device can be connected. I have had a few requests for 2 at once.
I can look at this, but I have a backlog of changes to get through first.

sebaone84, 2025-03-20:
Thank you very much.
I'll wait.
I bought the chocolate plus because I thought I could control the 20 songs with 5 banks and use bank 6 for the 4 effects, but I have to use banks 6 and 7 because I have to use A for on and B for off.