Builty/TonexOneController #114: chocolate connects but doesn't work, same as changes from screen

50pln, 2025-03-24:
in beta version 5 chocolate doesn't work for me, it's supposedly connected but it doesn't change anything

manual change from screen, also does not work. return to 4

DrD85, 2025-03-24:
Same problem with wired footswitches. Switching effects works, but presets does not. If I engage the compressor it switches to bank 256? and stays there

Builty, 2025-03-25:
I've released beta 6, please try. I expect it to be fixed now.

The issue relates to the Tonex "state" data. It's a data chunk that reports things like which preset is active. And it gets modifed and sent back to change presets.
Previously, I was using an array, and counting backwards from the end to locate the values to change.
In beta 5 I changed it to copy the data to a structure, which allows much easier data manipulation.
It seems however that the state data is not of a fixed length. IK does some data escaping on values, depending on which colours are used for the leds.
So the fixed structure worked for my pedal but not for others.

I've reverted to an array and offsets, but now defined the offsets to make the code more maintainable.

Przester-MZ, 2025-03-25:
Now (beta 6 ) everything works great at least for me (chocolate switch + zero platform). 
Once again respect and big thanks for your work.

One idea for the future - global volume control (I don't know if it's just my mistake in the settings or there's a bug in Tonex, but it's always set to 8 after switching on

50pln, 2025-03-25:
They work fine! Thanks a lot for your work!

As for the global volume control, I didn't notice that unless I'm looking in the wrong place.

DrD85, 2025-03-25:
Beta 6 works like a charm with the added footswitches and the SX1509 PCB

closed 2025-03-25 by 50pln