Builty/TonexOneController #115: Tuner Calibration

benhohlfeld, 2025-03-28:
Hi,

really fantastic work the controller (I use the web gui with the s3 Zero)!

Would it be possible to add the Tuner Calibration? I really need it, and it seems to be, at least for me, the only thing wich is missing.

Thanks in advantage!

Ben

Builty, 2025-03-28:
Hi Ben.
Thank you. Tuner calibration as in the reference frequency thing? 440 Hz etc?
If it's that, then yep I can add it.

benhohlfeld, 2025-03-28:
Hi,

thats it! Thanks again!

Builty, 2025-03-29:
Coded, will come with V1.0.9.2.

[image]

benhohlfeld, 2025-03-29:
> Builty
>  left a comment
> (Builty/TonexOneController#115)
> Coded, will come with V1.0.9.2.
>
> image.png (view on web)

Thanks a lot! 

Ben Hohlfeld

50pln, 2025-03-29:
upload as beta ;)

Builty, 2025-03-29:
I can upload a build here if needed.
I don't really want to start a new beta test phase for 1.0.9.2 quite yet, its way too soon.

50pln, 2025-03-30:
If it's not a problem, please upload it for 4.3B Amp, I'll be happy to test it.

Builty, 2025-03-30:
4.3B amp skin below.
Ben if you'd like one for the Zero, let me know.

[TonexOneController_V1.0.9.2_preview_1_Waveshare_4_3B_Display_Amp_Skins.zip](https://github.com/user-attachments/files/19525143/TonexOneController_V1.0.9.2_preview_1_Waveshare_4_3B_Display_Amp_Skins.zip)

faso82, 2025-03-30:
Hi everyone!

The tuner is available???

50pln, 2025-03-30:
I checked it for a short while, it works fine for me. Once when enabling effects from the desktop it restarted and hung, but after restarting I didn't have this problem anymore.

benhohlfeld, 2025-03-30:
Yes indeed! This would be great. 

What would I have to do with all the files? Or do I still just need to flash the 4 files in the bin folder as I did the first time? I’m kind of new to the matter.

Thank you again!

Ben

Builty, 2025-03-30:
Ben, here is the Zero build. Follow the same instructions as you did before for the first time you programmed it. 

[faso82](https://github.com/faso82), I haven't found any way of switching the tuner on or off so far. This new part is the global setting for the tuning reference frequency.

[TonexOneController_V1.0.9.2_preview_1_Waveshare_Zero.zip](https://github.com/user-attachments/files/19527528/TonexOneController_V1.0.9.2_preview_1_Waveshare_Zero.zip)

benhohlfeld, 2025-03-31:
Hi Greg,

I flashed the 4 Files and it’s working, (Thank you again for the tuner calibration!), but I don’t find any amp skin; what did I miss? Do I need a display now? I’m using the browser right now.

Thanks for your reply

Ben

Builty, 2025-03-31:
Hi Ben.
The amp skins are currently only available on the Wavesare 4.3B display. You don't need this.
I am thinking about adding skins to the web UI but its a nice challenge.

50pln, 2025-04-08:
I put together the zero and the step-down converters, put them in a box :) it controls the second tonex

[image]