Builty/TonexOneController #116: Free assignment wired midi Prg change => Tonex Preset

RafGuad, 2025-03-31:
Hi,
would it be possible to add in the wired midi section option to freely assign any Tonex preset slot to one or more midi program changes (eg Tonex preset slot 1 : Prg change 4 and/or 5 and/or 7...,  not only  Preset 1; 2; 3 = Prog 1; 2; 3 ) ? 

This is because my  integrated fx and midi controller sends fixed program changes at every Fx patch change (it is not programmable), so it would be useful if there was the ability to freely map program changes to any  slot of the Tonex so that there does not necessarily have a Tonex preset change at every fx patch change from the wired Midi controller.

RafGuad, 2025-04-02:
You are right, the only solution is to have a grid with fixed unique Midi Program Change values ​​from 0 to 127 and Tonex Preset Slot numbers from 1 to 20 freely assignable (from a drop-down menu or with a constrained value) to each or more program changes.

An idea for the layout could be this:

[image]

Builty, 2025-04-03:
Thanks, yeah this makes sense. I have noted down to implement.

Builty, 2025-05-14:
Hi. Recently a contributor implemented a change that lets you change the preset order. E.g. requesting preset 1 can now load preset 10. 
This will come with V1.0.10.2. Do you think this will meet the requirement?

Just in general, I have a lot of people asking for a lot of different things. It's becoming more and more difficult for me to keep everybody happy.

RafGuad, 2025-05-14:
Hi, I can imagine how many requests you get and everyone would like something. 

I'm sorry, but changing the preset order would not solve what I believe is a more general problem with any pedalboard that sends Program Change messages corresponding to the device's preset number from 1 to 127 without the possibility of remapping them. This happens for example with the Nux Cerberus, with the Line6 PodGo and I imagine with many other mid-range Midi controllers. 

It would therefore be useful if there was the possibility of freely mapping the Program Changes to any slot of the Tonex, so that there is not necessarily a Tonex preset change at every program change from the wired Midi controller. 
In this way all 20 slots of the Tonex are available, each with a different profile and associable to any Midi Program Change. 

In essence, the ability to remap the Midi Program Changes to Tonex Slot is necessary and would make the system much more flexible and integrable with any wired Midi controller without losing the number of slots available on the Tonex (if you do not want to change profile at every program change you just have to load the same profile in more slots of the Tonex).

Builty, 2025-05-23:
Sorry for the slow reply, I understand what you are saying. I've added this to the task list.

RafGuad, 2025-05-23:
Thanks a lot Greg, I'm sure that with this feature this project will be integrable with any midi controller with maximum capabilities.

Builty, 2025-09-30:
Hi. I agree and I can see it would be valuable. I just have a big list of things I would like to do.
I am keeping this in mind and will try to squeeze it in.

RafGuad, 2026-01-01:
Hi Greg, Happy New Year .
Is there hope for this midi implementiaion ? 
Thanks.

Builty, 2026-01-05:
Hi. I'm sorry that has gone on for so long without being done.
I have done a lot of web changes lately that make this easier now.

I am currently on holidays from work and doing some projects on my house, but I will probably have some time soon anyway as it will be too hot to work outside later this week.

Builty, 2026-01-09:
I have this almost done now. I should be able to do a beta in a few more days.

[image]

Builty, 2026-05-09:
Thanks guys, I will investigate and debug this one.