Builty/TonexOneController #120: High current consumption with Wifi On - ESP32 S3 zero

RafGuad, 2025-04-01:
Hi ,
from a measurement from 9V source to 5V DC regulator ( non switching ) + ESP32 S3 - zero  with Wifi On at 25% of the power, 
the current absorbed is quite high, about 650mA ( 220mA with wifi Off ), which added to that needed for the Tonex via USB C about 200 mA) becomes about 900mA ( almost 1 A ). 

However, the ESP32 with Wifi active has a considerable consumption if we consider that common pedalboard power supplies have at source  9V outputs max of 300-500mA. 

Couldn't the consumption of the ESP32 be optimized by further reducing the Wifi power to 10 - 15% ?

Builty, 2025-04-01:
Refer to the table below. Enabling WiFi incurs around 90 mA for receive, plus up to around 300 mA during transmit. At 25% TX power this will be lower, but I would not expect it to be linear. E.g. 25% power is probably not 25% of 300 mA.

Dropping to 10 -15% probably save a small amount, but it may end up with issues in connectivity.
It's likely also that enabling WiFi prevents the ESP32 core 0 from sleeping as much, which would cause additional current draw.

Thanks for the access point bug report, yes you are correct and I will fix that.

[image]

RafGuad, 2025-04-02:
From a measurement with Wifi active at 25% power, I detected about up to 600mA, instead with Wifi inactive about 220mA. 
So it would seem that the RF absorption data reported in the table ( 90 mA for RX + up to around 300 mA for TX) are to be added to the current required by the CPU ( about 220mA ) ?

The total absorbed current seems rather high compared to the data reported in the table, 
are we sure that the RF power % reduction option really works and that the other core processes remain idle ?

Builty, 2025-04-02:
Yes the figures in the table are just for the radio part itself, not for the entire chip.

The ESP32-S3 is a dual core micro. Core 0 is dedicated to WiFi/Bluetooth and whatever else Espressif have chosen to run the system. 
I can't control what happens there. I can only speculate that the increased processing required by WiFi would have an impact on the core's ability to sleep.

Builty, 2025-04-03:
You are right, it is an important consideration. 

A 7805 linear reg is not as efficient as a switching reg. It will have losses that end up as heat. They are generally around 40 to 70% efficient, where a switching reg is often in the 90% range.

In my day job as an Engineer, it would be rare that I would select a linear reg. Especially for dropping 4 volts at this kind of current. 

https://github.com/Builty/TonexOneController/blob/main/HardwarePlatforms.md#9v_power

This switching reg board came from Amazon and was very cheap and easy to use. I would suggest if minimal current draw is an important considation, you would be much better off with one of these.

RafGuad, 2025-04-04:
Yes, a switching converter is definitely necessary. I would have preferred a linear regulator with Tonex to limit any noise problems, the drawback of switching modules is the ripple and residual harmonics. The module you recommend seems to be among the many with the best features for the project, the only thing that leaves me doubtful is its miniaturization for which I fear heat dissipation and reduced effective current.

[image]

Builty, 2025-04-04:
I have a few of the first ones. I find it works well, and runs very cool. I can recommend it,
The second one is probably OK also but I have not tried it.

closed 2025-04-08 by RafGuad

RafGuad, 2025-04-08:
Very good with the first one, it works very well for the project, it has good efficiency and a minimal residual noises. It must be considered that the ESP S3 Zero with active wifi has a non-negligible absorption if added to that needed to power the Tonex, ultimately a 5v switching converter is necessary if you want to use the same 9v power supply in a pedalboard.