Builty/TonexOneController #123: WiFI access point terminates after disconnection

madejm, 2025-04-02:
When a device disconnects from Tonex Controller AP it terminates the access point and reconnecting is not possible unless the ESP is restarted.

In `wifi_config.c` function `wifi_event_handler` for case `WIFI_EVENT_AP_STADISCONNECTED` calls `wifi_kill_all();`. Could that be omitted to allow for reconnecting?

madejm, 2025-04-02:
I commented the line `wifi_kill_all();`and flashed my Zero. WiFi can be disconnected and reconnected multiple times, so it seems to work correctly, but I lack experience to tell if that behavior does not cause any other problems.

Builty, 2025-04-03:
Hi. Yes, sorry this is a bug for sure. My intention was that for timed mode, it was really for just doing the configuration, and then either disconnect or save/reboot when finished.

I should not be calling the kill on disconnect if its not timed mode however.
I will get this fixed soon. Thats for the assistance.

Builty, 2025-04-03:
Fix committed now. If using access point timed, it will kill the WiFi on disconnect. I think that fits in with its intended short term usage.
In access point (not timed) the WiFi is left enabled.

madejm, 2025-04-03:
Great!

closed 2025-04-03 by madejm