Builty/TonexOneController #124: Changing preset and global settings in Web Configuration forces Stomp Mode

madejm, 2025-04-02:
I'm currently using ESP32 Zero with Tonex One in Preset Browsing Mode to toggle between 3 presets.

When in Preset Browsing Mode or Dual Mode in Web Configuration a preset is selected from the dropdown menu, it always changes the preset in the Stomp Mode slot and puts the device to Stomp Mode.
Would it be possible to modify current slot preset instead of always Stomp Mode one?

Same with changing global BPM, it always switches Tonex to Stomp Mode. Would be great if we could prevent changing the device mode here too.

madejm, 2025-04-02:
With proper argument passed to `usb_tonex_one_set_preset_in_slot` saving to slots A/B/C works properly, but after changing preset or global param Tonex seems to always disable Preset Browsing mode. It works like this in Tonex Editor app too.

Builty, 2025-04-03:
At the moment I am forcing Stomp mode in the code. This is because I think it's the only practical solution for reliably achieving switching of the 20 presets. Before this change many people had it in A/B mode and got confused.

This code part is below. If you wish to make this work for A/B mode, it would be needed to change the 1 to a 0, or even remove this line totally, which would then leave the pedal in its current state.

[image]

madejm, 2025-04-03:
Yep I got that figured out. Thanks! Do you have any idea if different offsets in `PedalData.StateData` could control "Preset Browsing Mode"?

madejm, 2025-04-04:
I dumped the contents of `StateData` and there is no change when entering Preset Browsing mode. So it's always disabled by Tonex One internally when changing state data.
Thanks for help in understanding this one!

closed 2025-04-04 by madejm