Builty/TonexOneController #127: ESP32-S3 and Tonex One power source

johnk877, 2025-04-08:
I have a question - What happens if I plug in the Tonex One 9v dc power source and the USB-C port into the Tonex Controller (Wave Zero)? Will the Tonex One automatically prioritize drawing power from the 9v dc source instead of through the USB-C port?

Asking as I have cramped 2 wave zeros into a small enclosure and need to manage heat concerns.

Builty, 2025-04-08:
Hi John.
I've recently been made aware that its fine to use 9v power along with USB power. The pedal does favour the 9v. I don't think it would have a huge impact on the heat however. The Zero is just passing the 5v straight through.

To manage heat, the Zero does benefit from having a heatsink attached to the ESP32 chip on it. Something along the lines of this:
https://www.amazon.com.au/Aluminum-Heatsink-14x14x6mm-Conductivity-Maximizes/dp/B0DNFDMGX4

I haven't check that exact one for sizing, but that kind of thing that just sticks on. It helps the chip cool a lot better than without.

johnk877, 2025-04-08:
Thanks will try it out. Have another question on wired midi. The Adafruit board costs an arm and leg where I live, is it possible to connect the 3.3v, GND and data pin 5 directly to a TRS jack (with resistors) and use it as a TRS MIDI in?

closed 2025-04-08 by johnk877

RafGuad, 2025-04-15:
Hi,
for the Adrafuit you can build a clone yourself...
[
](https://diyelectromusic.com/2022/08/29/3v3-midi-module-pcb/)