Builty/TonexOneController #128: Wired midi help

johnk877, 2025-04-08:
Have another question on wired midi. The Adafruit board costs an arm and leg where I live, is it possible to connect the 3.3v, GND and data pin 5 directly to a TRS jack (with resistors) and use it as a TRS MIDI in?

Builty, 2025-04-08:
Hi John.
I know of another user who is doing this, but it is not what the Midi spec calls for. The spec requires an optocoupler for isolation.
It may work in some circumstances but there could be edge cases where it doesn't.

Is there any other Midi adaptor board available where you live? I did try the common Arduino one but had no luck getting it going. Some weird grounding issue.

RafGuad, 2025-04-08:
Hi, for the same cost reason I built myself an exact clone of the Adafruit kit, minimal expense and it works perfectly. As soon as I can I will post the schematic here ...

johnk877, 2025-04-09:
Thanks for the advice. The schematic will be super helpful!

johnk877, 2025-04-09:
Does a board like this work? [Midi shield](https://forum.arduino.cc/t/midi-shield-breakout-board/1191325/2)

Builty, 2025-04-09:
Regarding that Midi shield, when I said "I did try the common Arduino one but had no luck getting it going. Some weird grounding issue" this is the exact board I was referring too.

I found that if I had the board power supply connected to my PC ground (e.g. via USB-C) then the Midi would work OK. But if powered off an isolated supply, it would not work. The board has no schematic available and basic measurements with a multimeter didn't reveal why this occured. I ended up giving up on it and the Adafruit one worked perfectly first time and is properly documented.
The Midi shield was also quite poorly made. The big switch on it was full of crap and didn't make contect until I sprayed it with contact cleaner and cycled it a number of times.

RafGuad, 2025-04-09:
Here are all the instructions to self-build a 3V3 MIDI In / Out Module , 

https://diyelectromusic.com/2022/08/29/3v3-midi-module-pcb/
https://diyelectromusic.com/2021/02/15/midi-in-for-3-3v-microcontrollers/

It's a clone of the Adafruit MIDI Feather, I made with a multi-hole PCB only the section for the Midi Input in the ESP S3 zero and it works great! 

I attach the original schematic of the Adafruit and the DIY project, they are practically identical.
If you want to share them it could be useful for anyone.

[adafruit-midi-featherwing.pdf](https://github.com/user-attachments/files/19672612/adafruit-midi-featherwing.pdf)

[midimodule-DIY sch.pdf](https://github.com/user-attachments/files/19672618/midimodule-DIY.sch.pdf)

johnk877, 2025-04-11:
Can I check what PC or CC messages should be sent out to change presets?

Builty, 2025-04-12:
To select a specific preset, you can send Program change with the preset number.