Builty/TonexOneController #129: ESP32 S3 ZERO VSCODE PROBLEM

eniovasconcellos, 2025-04-08:
Hi man!

Did you use vscode + platformIO to program s3 zero? I can´t configure the real size of flash memory! I tryed configure platformIO.ini to 4M even the menuconfig setting 4M but nothing changes. When compiling the log show HARDWARE...8M always. Can you help me friend please? The upload ocurrs but not in  the right way. I am using esspressif devKit S3 as board in the platformIO.

Builty, 2025-04-09:
Hi. I use VSCode, but not platform IO. It's written using ESP IDF framework V5.0.2. 
I have some basic info here:  **https://github.com/Builty/TonexOneController/blob/main/FirmwareDevelopment.md**

All of the config is already set in the sdkconfig.defaults files (a common one, plus one per platform.)

I think if you install ESP IDF, everything should work OK.

madejm, 2025-04-11:
For ESP32 S3 Zero use `partitions_zero.csv`, this should be updated in the development doc.

Builty, 2025-04-12:
I've done that now. That whole page needs some more updates to refect the changes that happened since it was written.

eniovasconcellos, 2025-04-28:
Thank you man! Now I am fighting with chatGPT to solve problems about tinyUSB HOST lib  for me to connect esp32s3 to zoom ms50g.