Builty/TonexOneController #131: Prest up with midi and external Footswitch

DrD85, 2025-04-12:
Hi,
maybe I'm too stupid, but I want to assign a preset up to one of the external Footswitch fx buttons. I put CC 87 and 0 - 127, but it does nothing. In the tonex manual, it says toggle. Any tip on how to do it, would be appreciated .

Builty, 2025-04-12:
I've got Midi CC 87 set to change preset upwards. I didn't bother to look at the Midi values. Each receive of that message should change the preset up by 1.
Maybe I missed something there. I can run some tests myself to check.

DrD85, 2025-04-12:
I‘ll try again later, but it didn’t work before. I’m building a tonex Board for a friend with the 1.69 Version. Is there a chance for the effect icons to indicate, if it‘s turned on or off from the big brother coming to the small screen?

Builty, 2025-04-12:
Regarding the 1.69" getting effect icons, yep its already coded and coming in the next release.
https://github.com/Builty/TonexOneController/discussions/121

DrD85, 2025-04-13:
I tested it and it looks beautiful, but I get some random restarts when using it. I think it's somehow connect to the wifi. When I connect my phone to use the web control, it restarts very often

Builty, 2025-04-13:
I haven't experienced this myself, but I also haven't undertaken extensive testing of it.
I have been changing the configuration data also. Maybe try a flash erase and reprogram?

I hope to finish the last round of changes this week for V1.0.9.2, and then I'll be doing more testing and a beta phase again, so if this is an issue it will be found soon.

DrD85, 2025-04-13:
Thank you again. I will check, if it helps to erase the flash and try again with preset up tomorrow. Thank you again for your great work and I'm always happy to try betas

DrD85, 2025-04-14:
I tried it the Footswitch up CC87 0-0, CC87 0-127 etc and it's not working for me.
I also tried to erase the flash memory and flash the versions with the icons on the 1.69 and it still keeps on reebooting. I have no problems with the 1.0.8.2. The reeboots are more often, when I use the webconfig, so maybe it has to do something with it

Builty, 2025-04-14:
OK, I'll get  beta release out when I can. few more things to sort out

Builty, 2025-04-15:
Just confirming, yes the footswitch CC86 and 87 don't work. I will debug.

Builty, 2025-04-15:
Here's a preview build of 1.0.9.2, with the new UI icons and the above CC86/87 issue fixed.
Watch out however, I have done a lot of work in the footswitch configuration area. 
8 FX buttons for external switches.
1x2 and 1x3 layout support, and also a disabled option.
Plus changed the onboard footswitches to use the same layouts (up to 4 switches max) plus up to 4 FX switches also.

I had to adjust some indexes, you might need to set up yours again.

[image]

[image]

[TonexOneController_V1.0.9.2_preview_3_Waveshare_1_69.zip](https://github.com/user-attachments/files/19755578/TonexOneController_V1.0.9.2_preview_3_Waveshare_1_69.zip)

DrD85, 2025-04-22:
Everything works very good with beta 3

closed 2025-04-22 by DrD85