Builty/TonexOneController #133: Problem with running Controller build from sources on 4.3B

madejm, 2025-04-14:
Hi @Builty 

I have been successfully running the project built from source code on Zero board, but i have problems when running on 4.3B. **The device reboots every ~30 seconds**.
I'm using Vscode:
ESP-IDF v5.1.6
Flash method UART
Device target ESP32s3

Config:
```
Waveshare 4.3B
Amp Skins
LCD Display - ON
Small LCD - OFF
FS directly connected to GPIO - OFF
Double Frame Buffer - OFF
Bounce Buffer - OFF
Avoid tearing - ON

Custom partition table: partitions.csv
Offset of partition table: 0x8000
Generate MD5 - ON

Flash SPI: QIO
Flash size: 8MB

SPI RAM config: Octal Mode PSRAM

CPU frequency: 240MHz
```

Are there any other settings that need to be changed? I tried latest code on `main` as well as `V1.0.82` tag, both with same effect.

Builty, 2025-04-15:
Hi. It looks like you have them all, but its easier to use the VSCode project feature to manage this, along with the sdk config system.

Each platform has its own sdkconfig file, e.g. sdkconfig.ws43B_as.
And sdkconfig.defaults has common config.

The file esp_idf_project_configuration.json is used to tell the compiler to include both the defaults and the platform config, which takes care of all the config automatically and allows easy switching of platforms.

Reboot every 30 seconds is certainly strange. For that kind of issue its very handy to have the ESP32 serial debug output, but thats not easy on the 4.3B. I use 4.3 (not B) for that kind of dev, with dual USB.

madejm, 2025-04-15:
I used config for WS43B_as but got same results. What version of ESP-IDF are you using?

madejm, 2025-04-15:
Yes, that did the trick. Thanks 😉 Apparently version 5.1.6 has troubles with setting IDF target.

closed 2025-04-15 by madejm