Builty/TonexOneController #139: Screen related issues

madejm, 2025-04-16:
Things I have noticed that occur for 4.3B board on `1.0.8.2` release and current `main` code.

## Display blinking
Sometimes screen turns off for a split second. But everything works properly, so not a critical issue.

## Display goes off 
I encountered an issue that appears after a while of running the controller (sometimes after a minute, sometimes 10 min) and one of two things happen:

- the screen goes blank and the backlit is flickering; the footswitches **continue to work**
- the screen goes black; the footswitches **are not working**; the device does not reboot itself

In both cases the device needs to be restarted to work properly again.

Builty, 2025-04-17:
Hi, With the display blinking, do you mean the flicker that occurs just after boot, and also when saving configuration?
That one is an ESP issue. Refer https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/lcd/rgb_lcd.html and option CONFIG_LCD_RGB_RESTART_IN_VSYNC.

I haven't seen the second two myself. I run the 4.3B about 2 hours a night for maybe 6 days a week.
Could you detail the configuration you are using? What is connected, type of power supply etc.
Maybe something is different that I'm not testing for.

madejm, 2025-04-17:
My setup is:

- [ESP32-S3 4.3 LCD Touch version B](https://allegro.pl/oferta/wyswietlacz-hmi-esp32-s3-z-4-3-wifi-can-rs485-17231243502)
- Software version with Amp Skins
- [Sparkfun SX1509](https://allegro.pl/oferta/sx1509-breakout-16-liniowy-ekspander-portow-i-o-14093184815) - 9 footswitches connected
- Power:
  -  2.1 DC jack for 9v from pedalboard power supply ([this one](https://harleybenton.com/product/powerplant-iso-10ac-pro/))
  - USB-C power delivery 9V module ([this one](https://allegro.pl/oferta/899-wyzwalacz-tester-ladowarek-qc-pdc004-usb-c-9v-13936391567)) for powering from mobile/laptop charger
  - both are connected in parallel (issue happens either when powered from pedalboard or USB)

External footswitch layout 1x6B (footswitches 1-6)
Effect footswitches:
- 7: modulation power
- 8: delay power
- 9: tap tempo

When the issue occurs, the flickering looks like this:

https://github.com/user-attachments/assets/b8332b3a-3461-4ff0-9bfb-090dbf616e57

I'm currently testing if it also happens when I disconnect the SX1509 board.
Next I will disconnect the 9V USB modle to see if it might cause this issue.

Builty, 2025-04-19:
That looks like a power issue to me.
Which out of the power supply are you using? Some are 250 mA, some are 500 mA, some are 450 mA.
Even 500 mA is on the edge of not being enough.
Ideally it would want to be 1 amp.

madejm, 2025-04-21:
I just played a concert on this one and worked no problem running from pedalboard power (6 hours with rehearsal and stuff). I rewired the power jack, so that the USB module is disconnected when plugging the 9V DC. The USB module probably does some weird power interference.

buzzyc, 2025-04-22:
Possible derail - but would you be willing to share your pedalboard design once you're happy with it? It is exactly what I was looking once Builty added SX1509 support.

madejm, 2025-05-05:
I tested it with a cheap 9V 1A switching power supply. Now the issue occurs even faster, less than a minute running. 1A should be more than enough juice. I think that the frequency of switching power supply has some negative effect on the I2C.

Builty, 2025-05-06:
How long is the cable between the ESP32 and the SX1509? I2C is sensitive to the bus capacitance, which depends on wire length and construction.
Generally that kind of error would result in a lack of comms, but incorrect reads are also possible.

When doing the RS422 Midi on the 4.3B, I did see some strange behaviour when using the screw terminals to get 3.3v, but it went away in the end. This might be related to what you are seeing. 
Although I never did much with USB power on the 4.3B, always used 9v.

It might be worth trying to add some capacitors where the screw terminals are, accross the 3.3v and ground. That might help.
And take care of where the wires going to the SX1509 are running. Not near mains power or similar noisy areas.
Shielded cable could help if its being induced by that means.

madejm, 2025-05-06:
The I2C wires are not that long. I'll try to put a capacitor on the 3.3v bus.

[image]

madejm, 2025-05-06:
Adding capacitors on 3.3 bus and VIN did not help. I tried big electrolytics and smaller polyester (since they tend to handle high frequencies better). However looking at ESP32 schematic there are already plenty of capacitors on the power busses that would already eat any ripples.

I also noticed that while the display goes off, touch panel still works: clicking where next/previous arrows usually are, correctly changes the preset.

Since I don't have tools to properly debug the code, I'll try to print some errors to the web console, maybe there is some code that causes the issue.

madejm, 2025-05-08:
I did more digging into the issue and probably fixed it.

- First I confirmed that disconnecting SDA and SCL from SX1509 fixes the issue. So there must be some interference in the I2C.
- Then I disabled SX1509 initialization and communication in the code but left the board connected. The issue appeared again. That kept me convinced that it's not a software bug.
- Finally I watched your video demoing the external footswitches and noticed that you used the other side of the SX1509 board for connecting the IO. So I resoldered the SDA and SCL wires to the other side.

And it seems to be stable for now! Very weird, since the terminals on both sides are supposed to be linked. Is there anything that I missed in the SX1509 documentation? Might also be just a faulty unit.
If this is not just a coincidence, maybe [HardwarePlatforms.md](https://github.com/Builty/TonexOneController/blob/main/HardwarePlatforms.md) should point out which side terminals should be used.

Builty, 2025-05-08:
My guess would be that its not so much the side of the PCB, but where the wires are run. Maybe the power wire was running parallel to a footswitch wire that induced some interference.

This kind of device is not normally that susceptible to interference. I think the 4.3B is a little more susceptable to it due to some internal reasons.
It seems to be fine in most cases and not a major issue.