Builty/TonexOneController #143: Problem with starting ESP32 S3 Zero

Slowspark, 2025-04-18:
Hi. I uploaded the firmware to the ESP32 S3 Zero board and powered it with 5V, but after connecting it to the Tonex One with a USB-C-USB-C cable, the blue diode in the Tonex lights up and blinks sporadically, and crackling sounds come from the connected speaker, I have no idea what's wrong, I uploaded the 2 latest versions of the software What could be wrong?

madejm, 2025-04-18:
If Tonex button flashes blue and the knobs have a pale blue color it probably means that the pedal does not get enough power. I had that issue with a Zero board, my 9 to 5 voltage converter was faulty and outputed only about 3v. 

Can you check with a meter what voltage is supplied to the ESP32?

Builty, 2025-04-18:
Also make sure the 5v supply can give at least 1 amp,

Slowspark, 2025-04-19:
Thanks, the power supply was too weak

closed 2025-04-19 by Slowspark