Builty/TonexOneController #151: ESP32 S3 Zero Midi question

Slowspark, 2025-04-24:
Hello. I wanted to ask if ESP32 S3 zero supports midi in, midi out or both?, it is not clear to me, regards

Builty, 2025-04-24:
Hi. The Zero (and all other platforms) currently support Midi in only. As in, a PC or other Midi controller can send messages to the ESP32 to command it to make the Tonex do things.

Currently no support for Midi out. I can't really think of many cases where it would be helpful. All platforms do have a transmitter pin allocated just in case, but right now there is no code driving that transmit.

Slowspark, 2025-04-24:
I wanted to connect the ESP32 S3 Zero to another processor equipped with a display, midi output or serial output with information about the current preset to make it possible

Builty, 2025-04-25:
I don't think Midi allows sending strings, so couldn't send the preset name.
For that kind of thing, it would probably be better to code in a custom serial output that is easily read by the other processor.

Builty, 2025-04-25:
Or you could make a WiFi connection like the web editor does. Using web sockets you can get a huge amount of info.

marcuslupinus, 2025-04-25:
> I don't think Midi allows sending strings, so couldn't send the preset name. For that kind of thing, it would probably be better to code in a custom serial output that is easily read by the other processor.

You can send string with MIDI through 'SysX' message since they are «manufacturer dependent». If I remember correctly, the MIDI standard defines the SYSx start and end values and maybe some kind of header but the content can be used for whatever.

Slowspark, 2025-04-25:
I didn't mean the name, the current bank and preset number would be enough, I could read midi without modifying the program, I know there is a serial output but I would have to modify the firmware

Builty, 2025-04-26:
what Midi commands would the controller send, to confirm the current preset and bank?

Slowspark, 2025-04-27:
I think the program change messages would be enough. Banks would be defined in the second processor depending on how many switches

> what Midi commands would the controller send, to confirm the current preset and bank?