Builty/TonexOneController #152: Help on assigning Preset to a Footswitch

DZYHRY, 2025-04-25:
I would like to seek help regarding Web configuration in assigning a Preset to a footswitch. 

I followed the below set-up using 1.69 Non touch and 4 footswitches.

[image]

The device powers on, wifi and bluetooth connects flawlessy.

However, as I'm new to this kind of thing, I do not know how to assign a preset to a footswitch.

When I click this box, I can only see one preset like this, which is the preset assigned when in 'Stomp mode'. Is that normal?

[image]

Also, what parameters should I place in the below settings.

[image]

Builty, 2025-04-26:
Hi. With the preset list, yes this is correct. This list shows all 20 presets, and allows you to select a specific preset.

With the footswitches, this area is a bit complicated and I need to enhance the documentation.
Firstly, you are running firmware version v1.0.8.2. This version only allows effect switching via "external" switches, being ones attached using the little SX1509 expansion board. 
To get effect switching on the 4 directly connected switches you are using, you need to update to the latest V1.0.9.2.

The newer firmware has an "Int" tab for setting the "Internal" (directly connected) footswitches.

Now there are 2 parts to this. Preset switching, and effect switching.
For preset selection, its not setting a footswitch to a preset directly. It's setting a configuration like 1 row or 3 switches, or 1 row of 4 switches, with banking down by pressing 2 switches at once.
For 4 switches, the options are next/previous, 1x3, and 1x4.

For the effect switching, you can assign a specific footswitch to toggle an effect. Next/previous is also possible, but I don't have support for directly loading a preset via a specific footswitch. I did this part the same as the big Tonex pedal, and via Midi it doesn't have a direct preset setting.
This is something I could look at adding though.

DZYHRY, 2025-04-28:
> Hi. With the preset list, yes this is correct. This list shows all 20 presets, and allows you to select a specific preset.
> 
> With the footswitches, this area is a bit complicated and I need to enhance the documentation. Firstly, you are running firmware version v1.0.8.2. This version only allows effect switching via "external" switches, being ones attached using the little SX1509 expansion board. To get effect switching on the 4 directly connected switches you are using, you need to update to the latest V1.0.9.2.
> 
> The newer firmware has an "Int" tab for setting the "Internal" (directly connected) footswitches.
> 
> Now there are 2 parts to this. Preset switching, and effect switching. For preset selection, its not setting a footswitch to a preset directly. It's setting a configuration like 1 row of 3 switches, or 1 row of 4 switches, with banking done by pressing 2 switches at once. For 4 switches, the options are next/previous, 1x3, and 1x4.
> 
> For the effect switching, you can assign a specific footswitch to toggle an effect. Next/previous is also possible, but I don't have support for directly loading a preset via a specific footswitch. I did this part the same as the big Tonex pedal, and via Midi it doesn't have a direct preset setting. This is something I could look at adding though.

Thanks for the very detailed response. Will try setting it up again and give update to you. Hopefully it will work this time 👍

DZYHRY, 2025-05-14:
Just an update, Updating all the firmware and software, and installing the v1.0.9.2 solved my problem...

On to the next mod...controlling two Tonex One with the same set of Footswitch hahaha

Big thanks...

closed 2025-05-14 by DZYHRY