Builty/TonexOneController #168: Deactivaitng Preset UP/DOWN with internal switches

faso82, 2025-05-03:
I've connected a female TRS jack to the board as an internal switch, and then plugged the Ampero Switch, assigning two different commands to the switch (Tap Tempo to the A and Delay ON/OFF to the B). The commands seems to work fine, but they are unusable, because the Preset UP/DOWN configuration cannot be deactivated. If I press the switch A on the Ampero Switch, the board takes the Tap Tempo and also always scroll to the previous bank... Am I doing something wrong? If not,  is it possible to work on it for the next release? THX

madejm, 2025-05-03:
You can set the internal Preset Switch Layout to Disabled. Doesn't that solve your case?

faso82, 2025-05-03:
No, unfortunately turning the internal preset switch layout to Disabled doesn't solve the problem...

Builty, 2025-05-04:
I ran a test on the current code, and for me with the internal preset layout set to disabled, it works as expected.
Did you definately press the save button?
And if you go back into the menu again, does it show as disabled?

faso82, 2025-05-04:
i've reinstalled the last firmware, and tried again, but the issue remains; I press the save button when I disable the internal preset layout and after the controller reboots, it's shown as Disabled, the CC command is correct, but the Ampero Switch continues to work as Preset Up/Down; I post some photos to show the connection and the web config menu
thanks for the help

[image]
[image]
[image]

madejm, 2025-05-04:
@Builty i think this is an error in `footswitches.c`:

```c
#if CONFIG_TONEX_CONTROLLER_HARDWARE_PLATFORM_WAVESHARE_43B || CONFIG_TONEX_CONTROLLER_HARDWARE_PLATFORM_WAVESHARE_43DEVONLY
    // 4.3B doesn't have enough IO, only supports dual mode
    FootswitchControl.onboard_switch_mode = FOOTSWITCH_LAYOUT_1X2;
#else
    // others, get the currently configured mode from web config
    FootswitchControl.onboard_switch_mode = control_get_config_item_int(CONFIG_ITEM_FOOTSWITCH_MODE);
#endif
```

On 4.3B it doesn't take the value from config, always sets to `FOOTSWITCH_LAYOUT_1X2`

Builty, 2025-05-04:
Thanks Madejm, yes that will be it. I used a different platform to test.
Fixed now. Here is a preview build @faso82 .

[TonexOneController_V1.0.10.2_preview_2_Waveshare_4_3B_Display_Amp_Skins.zip](https://github.com/user-attachments/files/20030398/TonexOneController_V1.0.10.2_preview_2_Waveshare_4_3B_Display_Amp_Skins.zip)

faso82, 2025-05-05:
Thanks @madejm and @Builty 
i'll try the preview build later today and will let you know

faso82, 2025-05-05:
@madejm @Builty the preview build works fine; the int switch now works properly as assigned and no longer changes the Preset UP/DOWN
THX a lot!