Builty/TonexOneController #175: Webconfig only display some presets name when in "Station" mode

jamkid, 2025-05-06:
The preset list displayed fine when using the controller in AP mode. 
However, when using in Station mode, only the top 5, and the last preset names are shown. 

This is on the 1.69 non-touch version.
Tried using different controllers but same problem persisted. 
Tried changing preset names, reorder some of them, but still the same. 

Changing presets by changing the selection still works though. 

[image]

madejm, 2025-05-06:
It's probably due to my changes in [Custom preset order mapping #156](https://github.com/Builty/TonexOneController/pull/156). I haven't tested it in the station mode.
It's probably a race condition in the initial syncing preset names.

Does the LCD display on 1.69 show all preset names or does this bug occur there too?

jamkid, 2025-05-06:
/

madejm, 2025-05-06:
@jamkid could you please try to run the code from a branch in my fork?
https://github.com/madejm/TonexOneController/tree/web-preset-names-fix

I think that should fix it, but I can't test it properly right now.

jamkid, 2025-05-07:
Ive just tested your code, but unfortunately the same problem persisted.
I'll do more tests later to try to pinpoint the issue

Builty, 2025-05-07:
I tested and can reproduce. Looks to me like its the below. The message send to the WiFi queue failed. 
Looks like WiFi task is busy with the station connection and not servicing the queue, causing it to fill.

I'll try some fixes.

[image]

madejm, 2025-05-07:
Oh, the queue was fed 21 messages with presets, that's why some data was lost!

What kind of TTL-USB emulator exactly are you using? Is it possible to use it with 4.3B too?

jamkid, 2025-05-07:
I can confirm that it is working on my side too. Thank you! 
I will close this ticket =)

closed 2025-05-07 by jamkid

Builty, 2025-05-07:
@madejm I have a few TTL to USB adaptors.
One using CP210 driver that my employer supplied.
The other is an FDTI:
https://ftdichip.com/products/ttl-232r-3v3-aj/

Just need to make sure its a version for 3.3v and not 5v (5v TX output may damage the ESP32)

The log UART pins are ESP32 pins 36 and 37. On the 4.3 (not B) these connect to an onboard UART to USB chip.
On the 4.3B, the go to the RS485 transceiver, and I also use the TX for Midi.
You should be able to get access to the logs by disabling Midi, and connecting the USB adaptor to the RS485 transmitter. 

Waveshare labelled it as RS485_RXD. Maybe attach a mod wire to R64 to grab the signal.
[image]