Builty/TonexOneController #176: M-vave chocolate - Multiple Midi CC

Ignperlup, 2025-05-08:
Hello!
I'm trying chocolate for turn on/off effects in the Tonex. I'm using the advanced mode in the chocolate to send MIDI CC messages to the tonex and is working fine after activate MIDI CC in the tonexcontroller. The problem is when in the chocolate advanced mode two steps I set more than one messages in a bank, the tonex only receive the first one.

I would like to have this working for example to activate/deactivate different mod effects since I need to send múltiple messages at once (enabale/disable mod, change the mod type effect)

Any idea about this?

Thanks!!

Builty, 2025-05-08:
Hi. I'm not sure whether to blame the Chocolate or the controller here.

Could you possibly post screen shots of the Chocolate config? I can then setup mine the same way and check what happens.

Ignperlup, 2025-05-09:
Hello, yes. This is what I tried to activate mod and change de mod type at the same time

[image]

Regards

Ignperlup, 2025-05-13:
Good morning @Builty 
Could you check this?

Thanks!!

Builty, 2025-05-13:
Hi. Yes sorry it has taken longer than I hoped. I have a backlog of tasks to get through. I can probably get onto this one tomorrow.

Ignperlup, 2025-05-13:
No problem! just want to know if it is clear with the screenshot i sent.

Regards

Ignperlup, 2025-05-15:
Thanks!! I'm using ESP32-S3-zero

Regards

Builty, 2025-05-15:
Please try this one.

[TonexOneController_V1.0.10.2_preview_3_Waveshare_Zero.zip](https://github.com/user-attachments/files/20227020/TonexOneController_V1.0.10.2_preview_3_Waveshare_Zero.zip)

Builty, 2025-06-13:
Hi. V1.0.10.2 is released already. That one fixed up most of the issues in this area.
Recently I found a few more cases that were not handled properly (e.g. a mixture of PC and CC in the same Midi message.)
This is now also fixed, in the upcoming V1.0.11.2 release. I don't have a timeline for it yet, but I can give you a preview build if you need it.

Builty, 2025-06-20:
Hi. With the 4.3B, do you have the required RS485 PCB as detailed here?
https://github.com/Builty/TonexOneController/blob/main/HardwarePlatforms.md#midi

Now V1.0.10.2 handled multiple CC in one Midi message, but it didn't handle mixed PC and CC arriving in the same Midi message.
Please try this beta version:
https://github.com/Builty/TonexOneController/tree/main/Releases/beta/TonexOneController_V1.0.11.2_preview_1_Waveshare_4_3B_Display_Amp_Skins.zip

This one can support a mixture of PC and CC in the same message. Note however, it looks like the Tonex pedal doesn't like to be sent a message to change a parameter when it is in the process of changing presets. Another user who tested this had this issue, even though the controller firmware did the right thing. You might need to send the CC messages first the PC last.

Pelleinkiostro, 2025-06-21:
Hi thanks for the reply, no I don't have the RS485 PCB, I connected it directly as in the photo, and I connected it directly to the chocolate with mini jack, but it seems that the problem is the chocolate, as I tried to always connect the Morningstar MC6 MKII with a trs cable and it works perfectly, no problem, in the same pedal at the second press I activate the chorus or delay as in the video and everything works, tonight I'll try the beta and I'll let you know if it solves the chocolate, but at this point I'll use Morningstar which works great.

Builty, 2025-06-22:
Upload a video? Are you wanting to show something not working?

Pelleinkiostro, 2025-06-22:
Ok