Builty/TonexOneController #179: How to add support for Hotone Ampero Control

jjh84, 2025-05-09:
hello.

Today I programmed the firmware on a 1.69” LCD device and confirmed that the connection with the tonex one works fine.

Unfortunately I don't have a Chocolate unit from M-vave, instead I have an Ampero Control from Hotone.

The Ampero Control is basically similar to M-Vave's Chocolate device. The problem was that I was unable to connect to the Ampero Control from the Tonex Controller.

I utilized WebConfig to enable Custom bluetooth and filled in the bluetooth device name, but it didn't actually connect.

(Aside from this issue, it seems that rebooting the device after writing the Custom bluetooth name reinitializes the name).

Fortunately, I can write programs using the C language, so I would like to add Hotone Ampero Control support to this project if possible, but I have a few questions as I have never done any bluetooth programming before.

1. in order to add a new bluetooth footswitch, do I need to reverse engineer all the bluetooth characteristics that the device uses?

2. can you point me to any knowledgebase you referenced to implement Chocolate support from M-Vave?

Thank you.

Builty, 2025-05-10:
Hi,
The Chocolate pedal uses pairing without any security. That's how it can be used really easily. It just needs to be switched on.

Devices like the Ampero however use secure pairing, which is currently not supported in the Controller.

I have this noted in the task list under Secure Bluetooth pairing:
https://github.com/Builty/TonexOneController/blob/main/TaskList.md

There should be no need to do anything with the characteristics. That is standard Midi and already coded.
The part that needs to be done is the secure pairing, and that most likely needs the web UI adapted to allow for starting and displaying a list of nearby devices, and selection of a device from that list to bond with.

As far as knowledgebases, I wasn't able to find anything specific. I followed the ESP IDF examples to get basic bluetooth device support going, and then looked at some general BT midi info plus used the phone app "LightBlue" to help identify what was required.

There is a fair bit of work in this job, but I have had quite a few requests for it and it will be  good worthwhile feature.

jjh84, 2025-05-10:
Thank you for your response.

So, I'll have to study up on Bluetooth's secure pairing.

Can I leave one question here regarding firmware development?

I created a development environment utilizing the ESP-IDF framework, and the documentation for this project has been very helpful.

I was able to do both the firmware build and the firmware programming inside VS Code, which is very convenient.

My question is, once the Tonex one controller firmware is up and running, the serial port I was using as a monitor disappears. How do you monitor the logs output via `ESP_LOGI` in such an environment?

My development environment is macOS. The serial console that appears when I reset the device by pressing the `boot` button is `/dev/tty.usbmodel1301`.

jjh84, 2025-05-10:
I can see why: it's initializing the USB to communicate with the Tonex one.

Since my goal is to develop for Bluetooth, would it be okay to comment out `init_usb_comms()` in `app_main` and test it?

```c
    // init USB
    ESP_LOGI(TAG, “Init USB”);
    // init_usb_comms();
````

Builty, 2025-05-10:
Hi, yes you are correct about the log output, and yes commenting out the USB init is one trick I use a fair bit.
When developing features that don't require the Tonex pedal, its very handy.

I also have a Waveshare 4.3 (not B) that has dual USB ports. This allows both the Tonex pedal and program/debug logs at the same time. It's not as good as the 4.3B but gets the job done for a lot of development.

I also have 3.5 mm stereo jacks on a few other boards, that I connect to a TTL UART to USB adaptor (3.3 volt version) which can also view the logs.

Also regarding Bluetooth pairing, the ESP IDF examples folder has helpful samples. In your case:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_security_client

jjh84, 2025-05-10:
After a few attempts, I was able to successfully connect to Ampero Control.

It seems that Ampero Control does not use the secure pairing just like the M-Vave Chocolate device.

The reason I couldn't connect before was because the `midi_control.c` implementation didn't pass the routine that compares device names.

This may also be an issue with Ampero Control only. 

In the `ESP_GAP_BLE_SCAN_RESULT_EVT` event of the `esp_gap_cb()` function, take the Bluetooth scan result and verify that the expected device is connected. Here, the actual length of the string pointed to by `adv_name` is different from the value of `adv_name_len` when Ampero Control was detected. 

For our experiment, we renamed the Ampero Control to `FootCtrl` and connected it, where `adv_name` points to the string `FootCtrl`, and the value of `adv_name_len` was `9`. 

The exact cause needs further analysis. For now, I made the following patch and tried to connect again, and it worked.

```c
git diff
diff --git a/source/main/midi_control.c b/source/main/midi_control.c
index 2f835a1..d160ed4 100644
--- a/source/main/midi_control.c
+++ b/source/main/midi_control.c
@@ -1398,6 +1398,7 @@ static void esp_gap_cb(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *par)
                 { event
                     if (remote_device_names_length > 0)
                     {
+                      adv_name_len = strlen((char *)adv_name);
                         // loop over list of enabled devices to connect to
                         for (uint8_t loop = 0; loop < remote_device_names_length; loop++)
                         {       

```

Once connected, the MIDI control was successful. That's awesome!

jjh84, 2025-05-11:
Yesterday's FootCtrl data size of 9 was a mistake - that was the advertised data size for my OPUS device. The actual Ampero Control was advertising a size of 20 bytes, which is larger than the length of the actual device name.

the following data is logged at `BTM_CheckAdvData()` function in `btm_ble_gap.c`.

```
I (4431) BLE: 15 09 46 6f 6f 74 43 74 72 6c 00 00 00 00 00 00 00 00 
i (4432) ble: 00 00 00 00 00 00 
```

- The first byte (0x15) is the full size of the corresponding Adv Data
- The second byte (0x09) is the type of the corresponding Adv Data
- The third byte is the Adv Data (FootCtrl) 

This means that there is a difference between the size of the data advertised by Ampero Control and the actual size of the data. It could be that Ampero Control is advertising some sort of buffer size, and the actual data is being delivered smaller than the buffer.

For OPUS pedals, `adv_name_len` and the actual length of the name match.

```
I (8637) GATTC_CLIENT: Searched Device Name Len 9, OPUS 8085
i (8637) gattc_client: opus 8085
i (8637) gattc_client: 
```

For Samsung Galaxy Watch, It has same issue with the Ampero Control.

```
I (9115) GATTC_CLIENT: Searched Device Name Len 22, Samsung(862L)
I (9116) GATTC_CLIENT: Samsung(862L)
```

Thus, the value of `adv_name_len` can not be trusted. Is it safe if we compare string without the size?

Builty, 2025-05-11:
Hi. I've had a really busy weekend, and haven't had time to check this in detail.
I would recommend however using the free app "LightBlue" to connect to the pedal and investigate its data.

Below are some screen shots from the Chocolate pedal. It shows exactly what is being advertised, and I found this data very helpful.

In general with comparing strings, there is the function "strncmp" is one that checks 2 strings up to a maximum length. 
Its also possible to use "memcmp" and treat the data as uint8_t instead of char. 
 
[image]

[image]

jjh84, 2025-05-20:
Sorry for the delay.

I connected to my Ampero Control using the LightBlue app you mentioned. The screenshot below is the advertisement data sent by the Ampero Control. As a reminder, I changed the device name of the Ampero Control to FootCtrl so that the Tonex one controller will automatically try to connect to it.

[image]

As you can see on the screen, the data shown in the Advertisement Data matches what I posted last time.

- The first byte (0x15) is the full size of the corresponding Adv Data
- The second byte (0x09) is the type of the corresponding Adv Data
- The third byte is the Adv Data (FootCtrl)

In conclusion, the length of the device name (0x15) sent by Ampero Control is different from the actual length. This is true even if I change the device name.

The following screenshot is when the device name is set to AAAA.

[image]

The following screenshot is when the device name is set to AAAAAAAAAAAAAA.

[image]

As you can see, the data size is always 0x15. It never changes.

Here is the `BTM_CheckAdvData` function, which is the function used to determine the value of the `adv_name_len` variable.

```c
UINT8 *BTM_CheckAdvData( UINT8 *p_adv, UINT8 type, UINT8 *p_length)
{ }
    UINT8 *p = p_adv;
    UINT8 length;
    UINT8 adv_type;
    BTM_TRACE_API(“BTM_CheckAdvData type=0x%02X”, type);

    STREAM_TO_UINT8(length, p);

    while ( length && (p - p_adv <= BTM_BLE_CACHE_ADV_DATA_MAX)) {
        STREAM_TO_UINT8(adv_type, p);

        if ( adv_type == type ) {
            /* length doesn't include itself */
            *p_length = length - 1; /* minus the length of type */
            return p;
        }
        /* skip the length of data */ p += length - 1; /* skip the length of data */
        STREAM_TO_UINT8(length, p);
    }

    *p_length = 0;
    return NULL;
}
```

This function subtracts 1 from the length field value in the advertisement data and sets the value in the `p_length` pointer. This is why `adv_name_len` does not reflect the actual length of the data.

In my opinion, when the Tonex One Controller compares device names, it would be better to remove the routine that compares the name length first.

```c
if ((strlen(remote_device_names[loop]) == adv_name_len) && (strncmp((char*)adv_name, remote_device_names[loop], adv_name_len)) == 0)
```

It's using `strlen()` to check if the length of the name registered in the Tonex One Controller and the scanned `adv_name_len` are the same. In this case, the actual string is the same, but the routine cannot pass because the `adv_name_len` is incorrect.

Therefore, it seems that the above routine can be solved by simply replacing it with using `strcmp()`. It seems that running `strncmp()` is semantically equivalent to `strcmp()` if the string lengths are the same.

```c
If (strcmp(remote_device_names[loop], (char *)adv_name) == 0)
```

Builty, 2025-05-21:
Hi. Thanks for the super detailed info. It sounds like you are correct here.
I'm getting swamped with a lot of requests. I will need some more time to properly analyse your response.

jjh84, 2025-05-23:
yep, no problem. thank you for the awesome project 👍

Builty, 2025-05-23:
OK I understand now. The Ampero pads the device name with nulls, that breaks the strlen check in controller.
I think it should look like below.
The strncmp() function will check each character for a match, stopping the check when either it finds a null character (the 0x00 Ampero uses for padding) or if it hits the end of the advertised length, whichever occurs first.

The "n" function is helpful to avoid issues if either string is somehow missing a null terminator.

Could you possibly try the below change and see if it fixes it? If so I will double check its OK on the Chocolate also, and then commit the fix.
Thanks for the great analysis, it really helps a lot.

[image]

jjh84, 2025-05-25:
I changed to the code you provided and experimented.

I found that it connected to Ampero Control well. I think it's correct to use `strncmp()` as you said, thanks!

jjh84, 2025-05-26:
I've installed beta3 on my waveshare 1.69 and tested it. it works great. awesome!

jjh84, 2025-05-26:
The issue has been resolved, so we'll close it.