Builty/TonexOneController #186: usb_host.h not found during linking

demym, 2025-05-17:
Hi i am trying to compile the project on VScode with ESP-IDF extension, but it fails:

[image]

Any idea what could be causing this ? Using ESP-IDF v.5.0.8.

Thanks very much in advance

Builty, 2025-05-17:
Hi. This often happens if you have the "esp32" chip selected, as it does not support usb host.
You need to make sure you have the target device set to "esp32-s3"

[image]

demym, 2025-05-19:
thanks very much ! changed to s3 and worked ! thanks once again and once again congratulations !